mesh-iit/creo2urdf #23: CREO2URDF – MVP-2 first step FIXME ❌

Nicogene, 2023-05-11:
> At the moment we are gathering the datums without the copy geometry data structure, typical of shrinkwraps. Shall we add the processing of copy geometries to the checklist?

Mick3Lozzo, 2023-05-17:
Created assembly `SIM_ERGOCUB_1-0_HEAD` inside repository \\\\cad-mechanics\\projects\\simulation\_model\\ergocub.

@Nicogene @mfussi66

Nicogene, 2023-05-22:
> Get the datums by processing the copy geometries

Today w/ @mfussi66 noticed that the features inside the copy geometry can be taken transparently (e.g. as axis or as csys), so we didn't need to do anything in this sense.

We started then modifying `creo2urdf` in order to handle the new assembly and right now we have to find a way to correctly express the transform `prev_link_H_link`.

See https://github.com/icub-tech-iit/creo2urdf/commit/4932222537d3368b40fe821728d53a5b3cf2c596

We noticed that the root_H_link is fine, but then composing them we get strange transforms (some w/ `nan`). Probably we are using the idyntree API in the wrong way. But we will see better in the next days.

cc @pattacini @traversaro

traversaro, 2023-05-22:
> [](https://github.com/icub-tech-iit/creo2urdf/commit/4932222537d3368b40fe821728d53a5b3cf2c596)
>
> We noticed that the root_H_link is fine, but then composing them we get strange transforms (some w/ [...]

See https://github.com/icub-tech-iit/creo2urdf/commit/4932222537d3368b40fe821728d53a5b3cf2c596#r114447105 .

traversaro, 2023-05-24:
>  Right now the priority is to get the prints from idyntree, redirecting the cerr/cout to string/file.

See https://github.com/robotology/idyntree/issues/1064 . As a first test we can try https://stackoverflow.com/questions/10150468/how-to-redirect-cin-and-cout-to-files , even if it is probably not clean as it interferes with the global process state.

traversaro, 2023-05-24:
> said that in the export of the urdf a check is performed on the axis of the joint if it is close enough to the reference frame of the child link (or parent?), probably it is the error we are getting.

See https://github.com/robotology/idyntree/blob/v9.0.0/src/model_io/codecs/src/URDFModelExport.cpp#L365 . See also other `[ERROR]` messages to get an idea of the error we could be getting.

mfussi66, 2023-05-24:
And indeed we have the error on the exporter, related to the first joint:

```
[ERROR] URDFModelExport: Impossible to convert joint SIM_ECUB_HEAD_NECK_1--SIM_ECUB_HEAD_NECK_2 to a URDF joint 
without moving the link frame of link SIM_ECUB_HEAD_NECK_2 , because its origin is 0.00486151 4.13255e-16 -0.28977
```

mfussi66, 2023-05-24:
By defining the joint origin as the position of the child coordinate system we were able to fix the error. Initially, we put (0 0 0) assuming that the origin would be taken from the required transform, see [here](https://github.com/icub-tech-iit/creo2urdf/commit/c88dcae5d6186808ef6dfc5e2ffc4d56e4902b32#diff-5c3c6fa626b8b267b3047f685936aa454ee8cf8eaa74131932bd09693ac131c2R366) . I'm not 100% sure that it is the intended usage, but the urdf is now valid. 

However, we still need to correct the wrong pose of some of the links and joints:

[image]

Mick3Lozzo, 2023-05-24:
> We noticed that we are missing the SCSYS for the first link (NECK\_1), this is required for us to correctly define the transforms from links. We are assuming that the first one is an identity but it is not what we actually have. @Mick3Lozzo could you add it to the asm?

Added `SCSYS_NECK_1`.

@mfussi66 @Nicogene

mfussi66, 2023-05-25:
Joint axes fixed, the (simple) issue was that they are expressed in the root frame:

[image]

mfussi66, 2023-05-25:
So in the urdf in the previous post there was a problem in the link transforms. Basically, whenever we retrieved them, they had an unexpected rotation around the Z axis of 180deg, compared to the measurement by hand inside Creo.

So after a lot of checks and verifications, I think we have a potentially working URDF of the head, with all the links and joints in correct order and placement:

[image]

# BUT

What fixed the problem?

Since we need the transforms as iDynTree structures, we made a deep copy to convert from `pfcTransform3D` to `iDyntree::Transform`, assigning each element of the first to the constructor of the second (see https://github.com/icub-tech-iit/creo2urdf/blob/master/src/creo2urdf/src/Creo2Urdf.cpp#L127-L130)
I was afraid that Creo might be crazy enough to index matrix elements as (col, row) instead of (row, col) (basically transposing the matrix).
After applying the switch, we have the correct model 👀

I'm fairly sure that there are no other errors in the code that compensate for this transposition, but it might be useful to make a test in an environment with only two coordinate systems, and see what the transform is when printed from the toolkit in (row, col).

traversaro, 2023-05-25:
Probably it is not a col/row problem, just a matter of parent_R_child used instead of child_R_parent (or viceversa)

mfussi66, 2023-05-25:
I suspected that too, but what worried me was the fact that a simpl retrieval of a transform would return only the rows and columns of the rotation matrix switched, compared to the program:

[image]

closed 2023-05-29 by Nicogene