mesh-iit/creo2urdf #28: CREO2URDF – Fix validation and create the MVP2 urdf kinematically consistent

Nicogene, 2023-05-25:
### Task description

We want to:
- [ ] Fix validation of urdf (see https://github.com/icub-tech-iit/creo2urdf/issues/18#issuecomment-1545471080)
- [ ] Export the urdf of mvp2, kinematically consistent

### Definition of done

The above list is completed

Nicogene, 2023-06-06:
Today I made some research on what `idyntree` provide in terms of urdf exportation.

Right now it seems that the model can be exported only in terms of joints and links, the sensors can be read from the urdf but not exported.
Moreover, these are the sensors available

[image]

Regular/RGBD cameras are missing right now.

This means for proceeding we need to make some changes in idyntree or handle them in a "handcrafted" way, making postprocessing on the resulting urdf.

The latter option seems a little bit cumbersome, and if we contribute to idyntree this could help us and someone else in the future if need the same features.

In particular, the required changes are:

- Extend the sensors supported also to cameras
- Make export the sensors to the urdf
- Add the possibility to add XML blobs.

cc @pattacini @mfussi66 @traversaro

Nicogene, 2023-06-07:
> I do you extract from Creo the transform between the world and the base link? 

In theory we are asking this

traversaro, 2023-06-07:
But if I am not wrong, you you are not passing the world_H_base to the `iDynTree::KinDynComputation` class.

fiorisi, 2023-06-08:
> > I do you extract from Creo the transform between the world and the base link?
> 
> In theory we are asking this
> 
> 
> we assumed it was respect the world, but maybe it is expressed in the component frame? or Asm frame? This is something to be investigated @mfussi66

If I am not mistaken, in the `2BARS` assembly the two frames are overlapped for both parts.
[image]

[image]

Nicogene, 2023-06-08:
> But if I am not wrong, you you are not passing the world_H_base to the iDynTree::KinDynComputation class

Yes it was indeed this problem! We fixed it in https://github.com/icub-tech-iit/creo2urdf/commit/60e31518370611c1428617045272faf84800ab42

In https://github.com/icub-tech-iit/creo2urdf/commit/f4c321c7d6b5f80877ee71c871ba2546a3122800 we added a prompt for each joint where specify the joint position, since we do not have access to the mechanism API.

We did this check 

We first set the angle in the constraint
[image]

We inserted the value in the prompt

[image]

We passed the joint values to the `KinDynComputations` but we got errors between what tells creo and idyntree.

There is the possibility that we are moving the joints in creo in the wrong way, for this we will ask help to the mechies.

Nicogene, 2023-06-08:
Ah we also noticed that the `neck_pitch` is not in 0

[image]

We would need to keep it in that position but "reset" it to 0, because since we do not have access to the mechanism api, when we create the urdf we make the assumption that all joints are in 0.

cc @Mick3Lozzo @fiorisi

mfussi66, 2023-06-08:
> We passed the joint values to the `KinDynComputations` but we got errors between what tells creo and idyntree.

This was partially caused by `setRobotState` accepting radians instead of degrees. I corrected it, I'll update this post with results.

mfussi66, 2023-06-08:
Looks like we're using the wrong root frame in Creo. 
In the assembly, we have a frame at the top level, called ASM_CSYS, that could be considered our would frame. Every part, also, has a default frame, called CSYS, that is aligned with ASM_CSYS.

Now if we move one of the joints, CSYS also moves of course.

When we retrieve the transform info from Creo, we ask the transform from CSYS to and, e.g., SCSYS_NECK_HEAD, and not from ASM_CSYS. We did not notice this because at rest CSYS and ASM_CSYS are aligned.

See in the picture below, in which our print matches the measurement. The print was the data we used for validation. 

[image]

It seems then to me that we just have to fix our transform retrieval.

Nicogene, 2023-06-09:
Alongside @mfussi66 today we managed to fix the validation when moving one joint in the chain (see https://github.com/icub-tech-iit/creo2urdf/commit/5404abbc2fd6d8e8bcc03066d0b0410b739a5f75)

[image]
[image]
[image]

(We had to put - because right now we cannot solve the ambiguity of direction of the axis)

When we tried the validation by moving more than one joint (in particular we moved the realsense tilt

[image]

[image]

We got these errors in the link position/orientation

[image]

cc @fiorisi @traversaro @pattacini

Nicogene, 2023-06-09:
In https://github.com/icub-tech-iit/creo2urdf/commit/db20709ad89d69187917e5db7dd1c910ecc88f70 we fixed the validation when moving multiple joints 🎉 

[image]

The problem was that we thought we got `asm_csys_H_csys` instead we were getting `csys_H_asm_csys`.
Moreover, now we do not have to put minus in the joint value, this means that the direction of rotation seems consistent between urdf and cad.

[image]

closed 2023-06-09 by Nicogene