mesh-iit/creo2urdf #35: CREO2URDF – Start investigating how to retrieve child/parent links from joints

Nicogene, 2023-06-22:
Right now we are traversing the chain assuming that the prev and curr links are the parent and child of the joint, but we would like to retrieve this info from the axis, in order to manage multiple kin chains.

Nicogene, 2023-06-30:
I started working on this branch https://github.com/icub-tech-iit/creo2urdf/tree/mvp2.5, for now it is still a WIP.

The idea is to somehow populate a map traversing the assembly, containing for each axis the child and the parent link name and then use it when create the urdf.

Nicogene, 2023-07-03:
Here is the PR:

- #36

Nicogene, 2023-07-03:
I had the chance to test the latest changes w/ the ergocub upperbody asm (see https://github.com/icub-tech-iit/creo2urdf/issues/33#issuecomment-1618205102) and right now here is the result:

[image]

The strange things is that the upperarm are for some reason missing in the urdf, it is under investigation.

Nicogene, 2023-07-03:
Printing the model to file seems to containe the upper arm links and linked joints:

```
Model: 
  Links: 
    [0] SIM_ECUB_ROOT_LINK
    [1] SIM_ECUB_TORSO_1
    [2] SIM_ECUB_TORSO_2
    [3] SIM_ECUB_CHEST
    [4] SIM_ECUB_L_SHOULDER_1
    [5] SIM_ECUB_R_SHOULDER_1
    [6] SIM_ECUB_L_SHOULDER_2
    [7] SIM_ECUB_R_SHOULDER_2
    [8] SIM_ECUB_R_SHOULDER_3
    [9] SIM_ECUB_L_SHOULDER_3
    [10] SIM_ECUB_L_UPPERARM    <-----
    [11] SIM_ECUB_HEAD_NECK_2
    [12] SIM_ECUB_HEAD_NECK_3
    [13] SIM_ECUB_HEAD
    [14] SIM_ECUB_REALSENSE
    [15] SIM_ECUB_R_UPPERARM  <-----
  Frames: 
  Joints: 
    [0] SIM_ECUB_CHEST--SIM_ECUB_L_SHOULDER_1 (dofs: 1) : SIM_ECUB_CHEST<-->SIM_ECUB_L_SHOULDER_1
    [1] SIM_ECUB_L_SHOULDER_1--SIM_ECUB_L_SHOULDER_2 (dofs: 1) : SIM_ECUB_L_SHOULDER_1<-->SIM_ECUB_L_SHOULDER_2
    [2] SIM_ECUB_L_SHOULDER_3--SIM_ECUB_L_UPPERARM (dofs: 1) : SIM_ECUB_L_SHOULDER_3<-->SIM_ECUB_L_UPPERARM   <-----
    [3] SIM_ECUB_CHEST--SIM_ECUB_HEAD_NECK_2 (dofs: 1) : SIM_ECUB_CHEST<-->SIM_ECUB_HEAD_NECK_2
    [4] SIM_ECUB_HEAD_NECK_2--SIM_ECUB_HEAD_NECK_3 (dofs: 1) : SIM_ECUB_HEAD_NECK_2<-->SIM_ECUB_HEAD_NECK_3
    [5] SIM_ECUB_HEAD_NECK_3--SIM_ECUB_HEAD (dofs: 1) : SIM_ECUB_HEAD_NECK_3<-->SIM_ECUB_HEAD
    [6] SIM_ECUB_HEAD--SIM_ECUB_REALSENSE (dofs: 1) : SIM_ECUB_HEAD<-->SIM_ECUB_REALSENSE
    [7] SIM_ECUB_CHEST--SIM_ECUB_R_SHOULDER_1 (dofs: 1) : SIM_ECUB_CHEST<-->SIM_ECUB_R_SHOULDER_1
    [8] SIM_ECUB_R_SHOULDER_1--SIM_ECUB_R_SHOULDER_2 (dofs: 1) : SIM_ECUB_R_SHOULDER_1<-->SIM_ECUB_R_SHOULDER_2
    [9] SIM_ECUB_R_SHOULDER_3--SIM_ECUB_R_UPPERARM (dofs: 1) : SIM_ECUB_R_SHOULDER_3<-->SIM_ECUB_R_UPPERARM  <-----
    [10] SIM_ECUB_TORSO_1--SIM_ECUB_TORSO_2 (dofs: 1) : SIM_ECUB_TORSO_1<-->SIM_ECUB_TORSO_2
    [11] SIM_ECUB_ROOT_LINK--SIM_ECUB_TORSO_1 (dofs: 1) : SIM_ECUB_ROOT_LINK<-->SIM_ECUB_TORSO_1
    [12] SIM_ECUB_TORSO_2--SIM_ECUB_CHEST (dofs: 1) : SIM_ECUB_TORSO_2<-->SIM_ECUB_CHEST
```

But then this is the resulting urdf:
```
<?xml version="1.0"?>
<robot name="ECUB_UPPERBODY">
  <link name="SIM_ECUB_ROOT_LINK">
    <inertial>
      <mass value="2.823751482824641"/>
      <origin xyz="0.0024748962432214403 -1.050794470993344e-4 0.023450368787955644" rpy="0 -0 0"/>
      <inertia ixx="0.004380033669258687" ixy="-1.4369039688385886e-6" ixz="1.4098441290530498e-5" iyy="0.005630606285299465" iyz="-2.3299340994588874e-5" izz="0.0036166810108083755"/>
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="sim_ecub_root_link.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="">
        <color rgba="0.5 0.5 0.5 1"/>
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="sim_ecub_root_link.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="">
        <color rgba="0.5 0.5 0.5 1"/>
      </material>
    </collision>
  </link>
  <joint name="SIM_ECUB_ROOT_LINK--SIM_ECUB_TORSO_1" type="revolute">
    <origin xyz="-2.23890229767886e-18 -5.551115123142428e-20 0.060500000000000095" rpy="0 -2.2470258116354473e-44 5.656623812198896e-30"/>
    <axis xyz="-1 5.656623812198896e-30 1.4239886692466155e-15"/>
[... 329 lines cut ...]
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="sim_ecub_l_shoulder_2.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="">
        <color rgba="0.5 0.5 0.5 1"/>
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="sim_ecub_l_shoulder_2.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="">
        <color rgba="0.5 0.5 0.5 1"/>
      </material>
    </collision>
  </link>
</robot>

```

@traversaro do you have any hint why they could be omitted in the exportation?

traversaro, 2023-07-03:
There is no joint connecting `SIM_ECUB_L_SHOULDER_2` and `SIM_ECUB_L_SHOULDER_3` (and `SIM_ECUB_R_SHOULDER_2` and `SIM_ECUB_R_SHOULDER_3`, so the traversal generated in https://github.com/robotology/idyntree/blob/master/src/model_io/codecs/src/URDFModelExport.cpp#L531 does not reach SIM_ECUB_L_SHOULDER_3 and its children. Probably we could add `Model::isValid` or `Model::isConnected` to ensure that a manually built model is well connected and to early catch problems like this one.

Nicogene, 2023-07-03:
> Probably we could add Model::isValid or Model::isConnected to ensure that a manually built model is well connected and to early catch problems like this one.

We are already calling `isValid` but probably we have to call also `isConnected`.

> There is no joint connecting SIM_ECUB_L_SHOULDER_2 and SIM_ECUB_L_SHOULDER_3 (and SIM_ECUB_R_SHOULDER_2 and SIM_ECUB_R_SHOULDER_3),

Good catch thanks!

traversaro, 2023-07-03:
> > Probably we could add Model::isValid or Model::isConnected to ensure that a manually built model is well connected and to early catch problems like this one.
> 
> We are already calling `isValid` but probably we have to call also `isConnected`.

Sorry, those methods do not exists. I meant "add" as in "implement".

Nicogene, 2023-07-03:
I have understood what is the problem here, right now we always considered that two link are connected by a (revolute) joint, and the assemble has been made on the rotation axis of the joint. 

For `SHOULDER_2` and `SHOULDER_3` unfortunately, the mechanism is different, we should consider a fixed joint between those
since that is the fake joint of the ft sensors.

cc @fiorisi @mfussi66 @pattacini @Mick3Lozzo

Nicogene, 2023-07-04:
Today I analyzed how was defined the constraints between links for defining the joints and I discovered:

[image]

In the assembly provided by @Mick3Lozzo the revolute joints are defined aligning axis (`ALIGN`) and the fixed joints are defined aligning csys (`CSYS`).
Speaking w/ @fiorisi, this is just a way for defining joints in creo, they can be defined also aligning planes, axis+planes, etc.
For this reason, he suggested changing the upperbody assembly in order to define joints using different techniques, in this way, we can test how `creo2urdf` generalize and handle all these possibilities.
This could be an activity that can be done next sprint/PI.
How the joint has been created could be in-influent if we could access via API the to the Mech constraint object (toolkit for mechanica license), but right now we are forced to assume that joints are defined in a certain way.

>In the assembly provided by @Mick3Lozzo the revolute joints are defined aligning axis (`ALIGN`) and the fixed joints are defined aligning csys (`CSYS`).

In any case, in order to reach the MVP 2.5 I took this assumption for granted, and voila the complete urdf of the upperbody of ergoCub

[image]

cc @maggia80 @traversaro @pattacini @mfussi66

closed 2023-07-05 by Nicogene