mesh-iit/creo2urdf #40: CREO2URDF – Fix axis names in ergoCub assembly

Nicogene, 2023-07-05:
As reported [here](https://github.com/icub-tech-iit/creo2urdf/pull/39#issuecomment-1621877502), some axis names have not the `L/R` prefix in the ergoCub simulation assembly, and this breaks the exportation of the urdf of the entire robot.

Here is the list of joints to be renamed:
- MIDDLE_DIST_JOINT_AXIS
- MIDDLE_PROX_JOINT_AXIS
- THUMB_DIST_JOINT_AXIS
- THUMB_PROX_JOINT_AXIS

(This list has to be double-checked)

cc @pattacini @Lawproto @fiorisi

Lawproto, 2023-07-06:
To sum it up, back in the days I thought it was a good idea to place the joint datum axes inside the thumb and middle finger themselves. Since the very same assemblies are used in the right hand too, it felt to me like getting two birds with one stone.

[image]

I now get that this gives problems in other fields, instead. So, I am proposing to copy the reported joint axes - and possibly the related planes too for easier tracking, even though not strictly required - to the subassembly of each hand, rename them with a proper `L/R` prefix, then delete the current ones.

cc @Nicogene @Mick3Lozzo

Lawproto, 2023-07-06:
From [our internal mech guidelines section](https://github.com/icub-tech-iit/cad-libraries/wiki/Prepare-PTC-Creo-Mechanism-for-URDF#additional-datums) I get:

> The CAD model shall be updated to contain also a series of datum geometries (axis, planes, frames) that will be used to generate the URDF model. Specifically, we have to create geometries that will be used to assemble the Mechanism model and define the corresponding joints and geometries that will be used to define the location of specific sensors of the robot.
> 
> **These datums have to be created in the smallest assembly that "represents" the joint:**

@fiorisi to me, the bold part is exactly what I did and what lead us to break the exportation. I suggest we integrate this with something like: _with the further constraint that each joint name shall remain unique inside the whole robot assembly._
What do you think?

Lawproto, 2023-07-07:
I implemented the due changes with the commit above.

[image]

Shrinkwraps for the URDF export will be dealt with in another issue, along with other ones I have to prepare.
I think we can close this one, if I understand the issue title correctly.

Nicogene, 2023-07-10:
Hi @Lawproto! I'm happy to hear that this has been fixed in the robot cad!
I would keep it open until also the simulation model has been fixed, because creo2urdf needs that

Lawproto, 2023-07-20:
https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/164#issuecomment-1642393466

cc @Nicogene

Nicogene, 2023-07-20:
The fix of the axis names solved the issue of the missing links:

- https://github.com/icub-tech-iit/creo2urdf/pull/39#issuecomment-1643652447

Closing, thanks @Lawproto !

closed 2023-07-20 by Nicogene

Nicogene, 2023-10-03:
Probably @fiorisi and @Lawproto are already aware of it, then it is just a reminder for me, the ergoCub 1.0 simulation model still have the clash of names of the axis of the fingers (they miss the L/R prefix) 

[image]

Then for this reason the `creo2urdf` cannot work.

Lawproto, 2023-10-03:
> Probably @fiorisi and @Lawproto are already aware of it, then it is just a reminder for me, the ergoCub 1.0 simulation model still have the clash of names of the axis of the fingers (they miss the L/R prefix)

@Nicogene I understood we talked about updating the 1.1 mode at best l and tapulling the 1.0 model with as less effort as possible, as a general rule of thumb.

Anyway, I discovered it is possible to avoid the pain of exporting the simp reps again and rename the Creo axes directly in the sim_model, instead. I can help you doing like this in the 1.0 model, if it is helpful for you. Should take little time to me.

[image]

Nicogene, 2023-10-04:
> Anyway, I discovered it is possible to avoid the pain of exporting the simp reps again and rename the Creo axes directly in the sim_model, instead. I can help you doing like this in the 1.0 model, if it is helpful for you. Should take little time to me.

Thanks @Lawproto, let's see if we need to regenerate the 1.0 model (see [comment](https://github.com/icub-tech-iit/ergocub-software/issues/179#issuecomment-1745156217), cc @traversaro), in case I will ask you help.

Lawproto, 2023-10-04:
@Nicogene I had some time to spare at the end of this day, so I renamed the axes in the proper links for the 1.0 too, anyway.

Nicogene, 2023-10-05:
Great! Thanks @Lawproto 🎸 

In fact now `creo2urdf` works just fine also with ergoCubV1, see

[image]

In 
- https://github.com/icub-tech-iit/ergocub-software/commit/d3405c29519ace2a1732d783079225c1007d348e

fixed the axis orientation after generation the urdf w/ creo2urdf, but unfortunately the simulation model needs other fixes in order to be usable for generating a valid urdf:

```
1/4 Test #1: ergoCubGazeboV1ConsistencyCheck .....***Failed    0.01 sec
ergocub-model-test error:r_index_dist got direction of  x -0.947223 y -0.290266 z -0.136064 instead of expected  x 0.947223 y 0.290266 z 0.136064
ergocub-model-test : base_link test performed correctly 
ergocub-model-test : sole are parallel test performed correctly 
ergocub-model-test : translation between link l_foot_rear and sensor l_foot_rear_ft is non-zero, test failed, see  https://github.com/gazebosim/sdformat/issues/130  for more details.
ergocub-model-test : childLink_H_sensorFrame.getPosition(): 6.93889e-18 4.16334e-17 0.0223
ergocub-model-test : translation between link r_foot_rear and sensor r_foot_rear_ft is non-zero, test failed, see  https://github.com/gazebosim/sdformat/issues/130  for more details.
ergocub-model-test : childLink_H_sensorFrame.getPosition(): 6.93889e-18 2.77556e-17 0.0223
ergocub-model-test : translation between link l_foot_front and sensor l_foot_front_ft is non-zero, test failed, see  https://github.com/gazebosim/sdformat/issues/130  for more details.
ergocub-model-test : childLink_H_sensorFrame.getPosition(): 4.16334e-17 2.77556e-17 0.0223
ergocub-model-test : translation between link r_foot_front and sensor r_foot_front_ft is non-zero, test failed, see  https://github.com/gazebosim/sdformat/issues/130  for more details.
ergocub-model-test : childLink_H_sensorFrame.getPosition(): 1.38778e-17 -4.16334e-17 0.0223
```

See:

- https://github.com/icub-tech-iit/ergocub-software/issues/74
- https://github.com/icub-tech-iit/ergocub-software/issues/146
- https://github.com/icub-tech-iit/ergocub-software/pull/158
- https://github.com/icub-tech-iit/ergocub-software/issues/106

Making it usable w/ `creo2urdf` is right now at low priority, but it is nice to know that `creo2urdf` is usable on both simulation models

cc @pattacini @mfussi66