mesh-iit/creo2urdf #41: Add feature of using Yaml configuration file

mfussi66, 2023-07-21:
The `simmechancs-to-urdf` package is able to read a configuration file in YAML format, which is filled with parameters to customize the generated urdf. For more information, see: https://github.com/robotology/simmechanics-to-urdf/blob/master/README.md

The aim of this issue is to track the progress made when developing this feature. Ideally, we would like a 1-to-1 feature parity with `simmechanics-to-urdf`.

mfussi66, 2023-07-21:
Activity started in https://github.com/icub-tech-iit/creo2urdf/tree/fix/41 . It was trivial to add the `yaml-cpp` dependency by downloading the static package with `vcpkg`, and loading the library.

Renaming the joints was very simple (see the video below in mujoco)
https://github.com/icub-tech-iit/creo2urdf/assets/38140169/cb54b964-10ed-424a-a168-c31f9aada8e3

At the moment, iDynTree does not provide a setter for the link name, so we will have to be more careful with them since they are use in multiple parts of the code.

mfussi66, 2023-07-21:
Here is the configuration file I used: [ERGOCUB_all_options.txt](https://github.com/icub-tech-iit/creo2urdf/files/12122576/ERGOCUB_all_options.txt)

It's the same as the one in ergocub-software, but the prefix of links and joints is `SIM_ECUB_1-1_`.

traversaro, 2023-07-21:
> 
> At the moment, iDynTree does not provide a setter for the link name, so we will have to be more careful with them since they are use in multiple parts of the code. 

Did you consider doing the renaming before creating the IDybTree::Model ?

mfussi66, 2023-07-21:
> Did you consider doing the renaming before creating the IDybTree::Model ?

Indeed I did, it's what I did with the joints, which also do not have a setter for the names. 
Of course, we will rename the links before creating the iDyn Model, it's just that the code uses in multiple points the link names in a way that it's not trivially modifiable, and that's why I did not do it straight away.

mfussi66, 2023-07-24:
Progress update: atm the plug-in can handle the following YAML tags:
- robotName
- rename: both joints and links
- linkFrames: aka the name of the coord sys associated to a link (previously hardcoded in c++)
- reverseRotationAxis
- color (because why not, it was easy)

Sadly now the validation button was broken in the process, but it worked only for the single head, so it should not be a problem. We shall use a more complex validation approach anyway.

Nicogene, 2023-08-09:
Today @mfussi66 and I aligned on the status of `creo2urdf` and here are the missing functionalities:

- [ ] `exportedFrames`
- [ ] `assignedMasses`
- [ ] `assignedInertias`
- [ ] `XMLBlobs`
- [ ] `sensors`
- [ ] `forceTorqueSensors`

Nicogene, 2023-08-10:
W/ https://github.com/robotology/idyntree/pull/1088 and https://github.com/icub-tech-iit/creo2urdf/commit/bdd10b8accc31ea14990e78643dc11ce9ccdf6f7 is now possible to append xml blobs inside the `<robot></robot>` tag.

Nicogene, 2023-08-11:
https://github.com/icub-tech-iit/creo2urdf/commit/f7f21fdfb7c74efa586517bc1c0797f3be687703 implements `originXYZ/RPY` and `scale` functionality.

Nicogene, 2023-08-11:
https://github.com/icub-tech-iit/creo2urdf/commit/128ea9835bbb6bcba113f17b00625edb5f93c2e5 add `assignedMasses` and `assignedInertias` functionalities

```yaml 
assignedMasses:
 torso_1   : 2.0645667

assignedInertias:
  - linkName: torso_1
    xx: 0.0074060860
    yy: 0.0059470733
    zz: 0.0031780111
```

```urdf
<link name="torso_1">
    <inertial>
      <mass value="2.0645667"/>
      <origin xyz="-0.009639973621975028 6.168494736287859e-4 0.08244231288506343" rpy="0 -0 0"/>
      <inertia ixx="0.007406086000000001" ixy="6.975203277646223e-6" ixz="-1.2586871224024975e-5" iyy="0.005947073300000001" iyz="7.227736063238349e-4" izz="0.0031780111"/>
    </inertial>

```

mfussi66, 2023-08-18:
To address the export of FT sensors, there are 3 ways:
1. Modify iDynTree to handle ft sensors export with all the parameters in the yaml, together with the related XML blobs
2. Use libxml2 directly in creo2urdf to create an XML blob for each ft sensor, and then pass them as XML blobs to iDynTree using the feature created by @Nicogene 
3. Create a hardcoded, already formatted string, and add the ft parameters to it, then dump it to iDynTree as XML blob

I chose the second option because it was a reasonable middle ground between the other two. Creating a hardcoded string would have been quick and easy, but using libxml2 features could be useful since they could be merged directly into iDynTree in the future. Modifying directly iDynTree felt complex, and I was not sure if I would have followed the best practices.

Nonetheless, the latest commit creates valid xml files out of the ft sensors info. Here are two exemplary results:

# l_arm_ft_sensor

## creo2urdf

```xml
<?xml version="1.0"?>
<gazebo reference="l_arm_ft_sensor">
  <sensor name="l_arm_ft_sensor" type="force_torque">
    <always_on>1</always_on>
    <update_rate>100</update_rate>
    <force_torque>
      <frame>sensor</frame>
      <measure_direction>child_to_parent</measure_direction>
    </force_torque>
    <pose>-1.35724e-05 4.67557e-05 -1.13886e-05 -3.14159 3.33067e-16 -1.5708 </pose>
    <plugin name="left_arm_ft_plugin" filename="libgazebo_yarp_forcetorque.so">  <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_arm_ft.ini</yarpConfigurationFile></plugin>
  </sensor>
</gazebo>
```

## ergocub urdf

```xml
  <gazebo reference="l_arm_ft_sensor">
    <sensor name="l_arm_ft_sensor" type="force_torque">
      <always_on>1</always_on>
      <update_rate>100</update_rate>
      <force_torque>
        <frame>sensor</frame>
        <measure_direction>child_to_parent</measure_direction>
      </force_torque>
      <pose>0.0 0.0 0.0 -3.1415923328972943 2.923917580150936e-08 -1.5707962214346844</pose>
      <plugin name="left_arm_ft_plugin" filename="libgazebo_yarp_forcetorque.so">
  <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_arm_ft.ini</yarpConfigurationFile>
</plugin>
    </sensor>
  </gazebo>
```

[image]

# l_foot_rear_ft_sensor

## creo2urdf

```xml
<?xml version="1.0"?>
<gazebo reference="l_foot_rear_ft_sensor">
  <sensor name="l_foot_rear_ft_sensor" type="force_torque">
    <always_on>1</always_on>
    <update_rate>100</update_rate>
    <force_torque>
      <frame>sensor</frame>
      <measure_direction>parent_to_child</measure_direction>
    </force_torque>
    <pose>-1.14492e-16 2.91434e-16 0.0223 1.37274e-16 -6.81744e-17 -2.0944 </pose>
    <plugin name="left_foot_rear_ft_plugin" filename="libgazebo_yarp_forcetorque.so">  <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_foot_rear_ft.ini</yarpConfigurationFile></plugin>
  </sensor>
</gazebo>
```
## ergocub urdf

```urdf
  <gazebo reference="l_foot_rear_ft_sensor">
    <sensor name="l_foot_rear_ft_sensor" type="force_torque">
      <always_on>1</always_on>
      <update_rate>100</update_rate>
      <force_torque>
        <frame>sensor</frame>
        <measure_direction>parent_to_child</measure_direction>
      </force_torque>
      <pose>0.0 0.0 0.022299999999999986 0.0 -0.0 -2.0943952105869315</pose>
      <plugin name="left_foot_rear_ft_plugin" filename="libgazebo_yarp_forcetorque.so">
  <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_foot_rear_ft.ini</yarpConfigurationFile>
</plugin>
    </sensor>
  </gazebo>
```

[image]

Remarks:
- The computation of the pose was done through iDynTree. There are some cents of mm discrepancy wrt the cad, which shows exactly zero
	- I think the computation is correct, but it might be necessary to double-check
	- We can do the computation directly in Creo (preferable in my opinion), but it requires some modifications to the main loop that it would be better to discuss f2f
- 	the `<sensor>` chunk is missing at the moment, WIP

mfussi66, 2023-08-18:
> the `<sensor>` chunk is missing at the moment, WIP

This was addressed in: https://github.com/icub-tech-iit/creo2urdf/commit/a8d9407ed09b51ea2cca2e7b1c87b6b610d16199

The latest commit (https://github.com/icub-tech-iit/creo2urdf/commit/4714498221d2745c8fddb1a4df75fa3b26ca5ec6) also creates the xml tree for the sensors which are not FTs.

For example, the lidar:

**creo2urdf**

```xml
	<gazebo reference="head">
		<sensor name="lasersensor_head" type="ray">
			<always_on>1</always_on>
			<update_rate>100.000000</update_rate>
			<pose>5e-05 -1.93855e-15 0.134802 9.72195e-15 5.33157e-15 3.16761e-15 </pose>
			<always_on>1</always_on>
			<visualize>true</visualize>
			<update_rate>40</update_rate>
			<ray name="head_laser_sensor">
				<scan>
					<horizontal>
						<samples>360</samples>
						<resolution>2</resolution>
						<min_angle>0</min_angle>
						<max_angle>6.28</max_angle>
					</horizontal>
				</scan>
				<range>
					<min>0.04</min>
					<max>5</max>
					<resolution>0.01</resolution>
				</range>
				<noise>
					<type>gaussian</type>
					<mean>0</mean>
					<stddev>0.005</stddev>
				</noise>
			</ray>
			<plugin name="ergocub_yarp_gazebo_plugin_laser" filename="libgazebo_yarp_lasersensor.so">
				<yarpConfigurationFile>model://ergoCub/conf/sensors/gazebo_ergocub_lidar.ini</yarpConfigurationFile>
			</plugin>
		</sensor>
	</gazebo>
	<sensor name="lasersensor_head" type="ray">
		<parent link="head"/>
		<origin rpy="9.72195e-15 5.33157e-15 3.16761e-15 " xyz="5e-05 -1.93855e-15 0.134802"/>
	</sensor>
```

**ergocub urdf**
```xml
	<gazebo reference="head">
		<sensor name="lasersensor_head" type="ray">
			<always_on>1</always_on>
			<update_rate>100</update_rate>
			<pose>4.99999999999997e-05 0.0 0.13480199999999992 0.0 0.0 -3.6732051038684153e-06</pose>
			<always_on>1</always_on>
			<visualize>true</visualize>
			<update_rate>40</update_rate>
			<ray name="head_laser_sensor">
				<scan>
					<horizontal>
						<samples>360</samples>
						<resolution>2</resolution>
						<min_angle>0</min_angle>
						<max_angle>6.28</max_angle>
					</horizontal>
				</scan>
				<range>
					<min>0.04</min>
					<max>5</max>
					<resolution>0.01</resolution>
				</range>
				<noise>
					<type>gaussian</type>
					<mean>0</mean>
					<stddev>0.005</stddev>
				</noise>
			</ray>
			<plugin name="ergocub_yarp_gazebo_plugin_laser" filename="libgazebo_yarp_lasersensor.so">
				<yarpConfigurationFile>model://ergoCub/conf/sensors/gazebo_ergocub_lidar.ini</yarpConfigurationFile>
			</plugin>
		</sensor>
	</gazebo>
	<sensor name="lasersensor_head" type="ray">
		<parent link="head"/>
		<origin rpy="0.0 0.0 -3.6732051038684153e-06" xyz="4.99999999999997e-05 0.0 0.13480199999999992"/>
	</sensor>
```

As we can see, they match pretty nicely.

traversaro, 2023-08-18:
Nice job!

mfussi66, 2023-08-22:
So, after fixing:
- small errors in the generated xml chunks
     - the parent tag in the ft was missing
     - the imus must have the attribute set to "imu" in the `<gazebo> <sensor>` group
- forcing the assignment of inertias like in ergocub000
- adding the read of limits from csv

we can run the robot in Gazebo:

[image]

Nicogene, 2023-08-24:
After a T2T review we merged the PR, this activity can be considered done 🎖️

closed 2023-08-24 by Nicogene