mesh-iit/creo2urdf #43: CREO2URDF – Add feature to use a Yaml configuration file – Stint 2

Nicogene, 2023-08-11:
This is the follow-up of #41.
Here are the remaining features to be covered (taken from [here]()):

```[tasklist]
### Tasks
- [ ] `root`
- [ ] `exportAllUseradded`
- [ ] `filenameformat`
- [ ] `filenameformatchangeext`
- [ ] `forcelowercase`
- [ ] `stringToRemoveFromMeshFileName`
- [ ] `assignedCollisionGeometry`
- [ ] `limits`
- [ ] Mirrored inertia parameters
```

mfussi66, 2023-08-24:
I would also add to the list the "damping" and "friction" factors for each joint, as they are beneficial parameters for simulation, and are present in other robots. They can be added quite easily to iDynTree and then edded to the exporter.

Nicogene, 2023-08-24:
Putting on hold since #45 has an higher priority

Nicogene, 2023-08-28:
In:
- https://github.com/icub-tech-iit/creo2urdf/pull/47

I added the `assignedCollisionGeometry` parameter, here is the resulting minContacts urdf:

https://github.com/icub-tech-iit/ergocub-software/commit/f7ce6c1e1b9f1c56fa745962091b0e83ead44733

Nicogene, 2023-08-28:
Chatting w/ @traversaro it came out that just one of `filenameformatchangeext` and `filenameformat` is sufficient, and the [`mirroredInertia`](https://github.com/robotology/simmechanics-to-urdf/blob/master/README.md#mirrored-inertia-parameters) parameter was a workaround for a bug in simmechanics link, since we don't use anymore simmechanics we should not need it.
In case we will implement it later on.