mesh-iit/creo2urdf #45: CREO2URDF – handle all the errors came out from the urdf unit tests

Nicogene, 2023-08-24:
In this branch of ergocub-software (https://github.com/icub-tech-iit/ergocub-software/tree/refactorRepo) I started refactoring the structure for hosting only the urdf and conf files for using creo2urdf, and running the unit tests w/ the urdf obtained from creo2urdf I got these errors:

```
ergocub-model-test error:l_shoulder_pitch got direction of  x -3.26128e-16 y 1 z 3.88578e-16 instead of expected  x 0.250563 y 0.935113 z 0.250563
ergocub-model-test error:r_shoulder_pitch got direction of  x -4.68375e-16 y 1 z 3.60822e-15 instead of expected  x -0.250563 y 0.935113 z -0.250563
ergocub-model-test error:r_thumb_add got direction of  x -0.116648 y 0.227771 z 0.966702 instead of expected  x -0.329847 y 0.160871 z 0.930227
ergocub-model-test error:r_thumb_prox got direction of  x -0.329847 y 0.160871 z 0.930227 instead of expected  x -0.525416 y 0.838737 z -0.143034
ergocub-model-test error:r_thumb_dist got direction of  x -0.439716 y 0.892571 z -0.0998355 instead of expected  x -0.525416 y 0.838737 z -0.143034
ergocub-model-test error:r_index_add got direction of  x -0.250563 y 0.935113 z -0.250563 instead of expected  x 0.250563 y -0.935113 z 0.250563
ergocub-model-test error:r_ring_prox got direction of  x 0.961047 y 0.271447 z 0.0520081 instead of expected  x 0.973762 y 0.226085 z -0.0259272
ergocub-model-test error:r_pinkie_prox got direction of  x 0.961047 y 0.271447 z 0.0520081 instead of expected  x 0.973762 y 0.226085 z -0.0259272
ergocub-model-test error:l_shoulder_yaw got direction of  x -0.116648 y -0.227771 z 0.966702 instead of expected  x 0.116648 y 0.227771 z -0.966702
ergocub-model-test error:l_wrist_pitch got direction of  x 0.250563 y 0.935113 z 0.250563 instead of expected  x -0.250563 y -0.935113 z -0.250563
ergocub-model-test error:l_wrist_yaw got direction of  x -0.116648 y -0.227771 z 0.966702 instead of expected  x 0.116648 y 0.227771 z -0.966702
ergocub-model-test error:l_thumb_add got direction of  x 0.116648 y 0.227771 z -0.966702 instead of expected  x 0.329847 y 0.160871 z -0.930227
ergocub-model-test error:l_thumb_prox got direction of  x 0.329847 y 0.160871 z -0.930227 instead of expected  x 0.525416 y 0.838737 z 0.143034
ergocub-model-test error:l_thumb_dist got direction of  x 0.439716 y 0.892571 z 0.0998355 instead of expected  x 0.525416 y 0.838737 z 0.143034
ergocub-model-test error:l_index_add got direction of  x 0.250563 y 0.935113 z 0.250563 instead of expected  x -0.250563 y -0.935113 z -0.250563
ergocub-model-test error:l_ring_prox got direction of  x -0.961047 y 0.271447 z -0.0520081 instead of expected  x -0.973762 y 0.226085 z 0.0259272
ergocub-model-test error:l_pinkie_prox got direction of  x -0.961043 y 0.271458 z -0.052021 instead of expected  x -0.973762 y 0.226085 z 0.0259272
```

and 

```
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame l_arm_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame r_arm_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame l_leg_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame r_leg_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame l_foot_rear_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame r_foot_rear_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame l_foot_front_ft as expected.
ergocub-model-test : model /home/ngenesio/icub-tech-iit/ergocub-software/build/ergoCub/robots/ergoCubGazeboV1_1/model.urdf does not contain frame r_foot_front_ft as expected.
```

The errors on the axis cannot be fixed reverting them, for the shoulder pitches there is something to be investigated, for the hands instead the discrepancy respect ergoCub 1.0 is due to the fact the hands changed? (@Lawproto @fiorisi)

The errors on the ft sensors are due to the fact the frames are not exported also if explicitly asked from the yaml (@mfussi66)

(For generating the urdf I used this [yaml ](https://github.com/icub-tech-iit/ergocub-software/blob/refactorRepo/urdf/creo2urdf/data/ergocub1_1/ERGOCUB_all_options.yaml) file)

Nicogene, 2023-08-25:
Alongside @mfussi66, today we started addressing the errors came out from the unit test, here is the PR:

- #46 

In particular, we fixed most of the axis issues by using the `root_H_parent_link` instead of `root_H_child_link`, given that 

[image]

Right now the only remaining faulty axis are:

```
ergocub-model-test error:r_thumb_prox got direction of  x 0.439716 y -0.892571 z 0.0998355 instead of expected  x -0.525416 y 0.838737 z -0.143034
ergocub-model-test error:r_thumb_dist got direction of  x -0.439716 y 0.892571 z -0.0998355 instead of expected  x -0.525416 y 0.838737 z -0.143034
ergocub-model-test error:l_thumb_prox got direction of  x 0.439716 y 0.892571 z 0.0998355 instead of expected  x 0.525416 y 0.838737 z 0.143034
ergocub-model-test error:l_thumb_dist got direction of  x 0.439716 y 0.892571 z 0.0998355 instead of expected  x 0.525416 y 0.838737 z 0.143034
```
Maybe it is due to a different rest pose? Or to the new kinematics? (cc @Lawproto @fiorisi)

It has to be referred into the parent link frame.

We managed also to export the frames of the sensors and ft sensor, but right now we are stuck on defining correctly their pose.

As pose we are setting `parent_link_H_ft_frame`:

[image]

```xml
<pose>-0.00435 -2.77556e-17 -0.0543 3.14159 2.498e-16 -1.5708 </pose>
```

But the "correct" pose should be according to ergoCubSN000

```xml
<pose>0.0 0.0 0.0 -3.1415923328972943 2.923917580150936e-08 -1.5707962214346844</pose>
```

The strange thing is the 0 0 0 for the traslational component, maybe we are missing something on how the ft sensor has to be defined? (@traversaro)

Lawproto, 2023-08-25:
> Right now the only remaining faulty axis are:
>
> ```
> ergocub-model-test error:r_thumb_prox got direction of  x 0.439716 y -0.892571 z 0.0998355 instead of expected  x -0.525416 y 0.838737 z -0.143034 [...]

I am not sure I understand the "faulty" part. but it seems to me that the software noticed that the angles of the thumb axis changed. Which is perfectly normal, since we changed the metacarpus angle as of our interactions with the HSP group.

[image]

traversaro, 2023-08-25:
> The strange thing is the 0 0 0 for the traslational component, maybe we are missing something on how the ft sensor has to be defined? (@traversaro)

I guess that pose is jointFrame_H_sensor, and jointFrame = childLinkFrame in this case.

Nicogene, 2023-08-25:
W/ https://github.com/icub-tech-iit/creo2urdf/commit/cfd18f64b81c86feaf11ae4e9e2fb17e95440856 I managed to correct the pose of the ft sensor as it should be, I had to increase the tolerance of this check 
```c++
checkTransformAreEqual(root_H_frame, root_H_sensor, 1e-5)
```
to `1e-5` because `1e-6` was triggering an error, even if the transform were identical when printing them 🤔 
I hope this tolerance is fine @traversaro 

Right now the unit tests are fine 🎉 

```
Test project /home/ngenesio/icub-tech-iit/ergocub-software/build
    Start 1: ergoCubGazeboV1ConsistencyCheck
1/3 Test #1: ergoCubGazeboV1ConsistencyCheck .....   Passed    0.03 sec
    Start 2: ergoCubGazeboV1_1ConsistencyCheck
2/3 Test #2: ergoCubGazeboV1_1ConsistencyCheck ...   Passed    0.02 sec
    Start 3: ergoCubSN000ConsistencyCheck
3/3 Test #3: ergoCubSN000ConsistencyCheck ........   Passed    0.03 sec
```

PR #46 (2023-08-25): Added exportation of ft sensors frames and fix axis definition
It fixes #45 

This PR  adds:
- The possibility of exporting frames for sensors and ft sensors
- Fix the axis definition
- Fix the ft sensor pose

See https://github.com/icub-tech-iit/creo2urdf/issues/45#issuecomment-1693382170
closed 2023-08-25 by Nicogene