mesh-iit/creo2urdf #48: CREO2URDF last step tbd

Nicogene, 2023-08-30:


Nicogene, 2023-08-30:
In [`89202a8` (#49)](https://github.com/icub-tech-iit/creo2urdf/pull/49/commits/89202a8b8ddf56078f6e9f888d667655f9b5bbb1) it has been added the possibility to open a dialog for selecting the yaml and csv from filesystem.

[image]

If cancel is pressed the plugin stops the execution.

Nicogene, 2023-08-30:
xref: 
- https://github.com/robotology/idyntree/pull/1094

pattacini, 2023-08-30:
>Add damping and friction to the joints

See:
- https://github.com/robotology/idyntree/pull/1094

mfussi66, 2023-08-31:
> xref:
> 
>    [Add export to urdf of joint dynamics: damping and static friction coefficients robotology/idyntree#1094](https://github.com/robotology/idyntree/pull/1094)

The related usage in creo2urdf is at the moment in my fork: https://github.com/mfussi66/creo2urdf/commit/292aa91cc5eb778d7419ce420468d7dd48e84cc6

mfussi66, 2023-09-01:
Now that the PR in iDynTree is merged, the changes can be brought to the plugin: [`04c8971` (#49)](https://github.com/icub-tech-iit/creo2urdf/pull/49/commits/04c8971501677d2821685c344f69fcc010961603)

```diff

    <joint name="torso_pitch"
           type="revolute">
        <origin xyz="-7.30886630439206e-17 -4.930380657631324e-32 0.11800000000000001"
                rpy="3.9719461100401744e-15 2.710505431213761e-20 1.0587911275090697e-22" />
        <axis xyz="1.058791184074231e-22 1 4.336808689942018e-19" />
        <parent link="torso_1" />
        <child link="torso_2" />
        <limit lower="-0.3141592653589793"
               upper="0.7853981633974483"
               effort="1e+9"
               velocity="1e+9" />
+        <dynamics damping="2"
+                  friction="0" />
    </joint>

```

PR #49 (2023-09-01): creo2urdf: add file dialog and damping/friction for the joint
It fixes #48

It adds
- File browsing
- Static friction and dumping according to urdf spec
closed 2023-09-01 by Nicogene