mesh-iit/creo2urdf #53: CREO2URDF – Add the possibility to run `URDF` generation from the terminal

FabioBergonti, 2023-09-08:
It would be nice to have the possibility to generate the `URDF` from the terminal to automate their generation.

Nicogene, 2024-01-24:
I found this interesting link:
- https://community.ptc.com/t5/Customization/How-to-start-a-batch-with-toolkit/td-p/499638

In particular we should call something like
```bat
call "%CREO_INSTALL_PATH%\Parametric\bin\parametric.exe" -g:no_graphics -batch_mode -i:rpc_input -RUN_MY_DESIRED_APPLICATION
```

I have to understand how to handle `-RUN_MY_DESIRED_APPLICATION`, how to pass to it the arguments(yml, csv) and how to open a specific assembly.

Nicogene, 2024-03-19:
The lack of an automatic generation from Github Action is becoming an issue since the number of maintained urdf is exploding:

- https://github.com/icub-tech-iit/ergocub-software/issues/228

Nicogene, 2024-04-05:
Habemus `creo2urdf` working from terminal! It is a first draft, the parsing is quite primitive, also because the Toolkit does not provide anything for parsing the parameters but today I managed to generate the urdf running this command line:

```bat
call "C:\Program Files\PTC\Creo 9.0.2.0\Parametric\bin\parametric.exe" -g:no_graphics creo2urdf -batch_mode -i:rpc_input +C:\Users\ngenesio\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm +C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml +C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_joint_all_parameters.csv +C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf

```

As you can see the `-` are used for passing the parameters to creo, the `+` are used for passing the parameters to the plugin, but then they are just plain C `argc, argv`, right now it is not very user friendly.

cc @pattacini @traversaro @FabioBergonti @mfussi66

traversaro, 2024-04-05:
Great!

pattacini, 2024-04-05:
Awesome!

Nicogene, 2024-04-05:
>

In:
- [`a9b4f9d` (#86)](https://github.com/icub-tech-iit/creo2urdf/pull/86/commits/a9b4f9d3924fdeee1f63b6c1841316417d999a39)

I added a `run_creo2urdf.bat` that allow me to print out how the plugin is launched
```bat
@echo off

ECHO Using asm  path: %1
ECHO Using yaml path: %2
ECHO Using csv path: %3
ECHO Using output path: %4

IF %1.==. GOTO No1
IF %2.==. GOTO No2
IF %3.==. GOTO No3
IF %4.==. GOTO No4

"C:\Program Files\PTC\Creo 9.0.2.0\Parametric\bin\parametric.exe" -g:no_graphics -batch_mode -i:rpc_input creo2urdf +%1 +%2 +%3 +%4

GOTO End1

:No1
  ECHO asm path not specified, exiting.
GOTO End1
:No2
  ECHO yaml path not specified, exiting.
GOTO End1
:No3
  ECHO csv path not specified, exiting.
GOTO End1
:No4
  ECHO output path not specified, exiting.
GOTO End1

:End1

```

It is lanched:
```
run_creo2urdf.bat C:\Users\ngenesio\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_joint_all_parameters.csv C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf
```

and here is the output:

```
Using asm  path: C:\Users\ngenesio\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm
Using yaml path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml
Using csv path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_joint_all_parameters.csv
Using output path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf
```

I think that there is plenty of things that we can do (for example the Creo path is hardcoded, we might use some env variable), but for now I will keep this new feature for developer only, we will improve it and then publicize later

cc @mfussi66 @pattacini @traversaro

Nicogene, 2024-04-05:
> > Moreover, I am using `ProTKPrintf` but nothing is diplayed on the windows command prompt, and it is not clear when the generation of the urdf finishes.
> 
> Maybe you can check in the .bat if the file model.urdf exists to check for success?

Good idea, I will chatgpt-it :D

Nicogene, 2024-04-05:
Following @pattacini advice:
- https://github.com/icub-tech-iit/creo2urdf/pull/86#issuecomment-2039772733

I converted it into ps script, and this allowed me to improve it a lot:
- I added a help section
- I added the await of the finish of the urdf generation
- I added names to the parameters
- I added a final check about the existence of `model.urdf` after the command exits

Here is the help printed:
```

NAME
    C:\Users\ngenesio\icub-tech-iit\cad-libraries\run_creo2urdf.ps1

SYNOPSIS
    Converter for Creo assembly to URDF format.

SYNTAX
    C:\Users\ngenesio\icub-tech-iit\cad-libraries\run_creo2urdf.ps1 [[-asmPath] <String>] [[-yamlPath] <String>] [[-csvPath] <String>] [[-outputPath] <String>] [<CommonParameters>]

DESCRIPTION
    This script converts  Creo assembly to URDF format using the creo2urdf plugin.

RELATED LINKS

REMARKS
    To see the examples, type: "get-help C:\Users\ngenesio\icub-tech-iit\cad-libraries\run_creo2urdf.ps1 -examples".
    For more information, type: "get-help C:\Users\ngenesio\icub-tech-iit\cad-libraries\run_creo2urdf.ps1 -detailed".
    For technical information, type: "get-help C:\Users\ngenesio\icub-tech-iit\cad-libraries\run_creo2urdf.ps1 -full".
```

Here is how it is launched now:

```ps
.\run_creo2urdf.ps1 -asmPath "C:\Users\ngenesio\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml" -yamlPath "C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml" -csvPath "C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_joint_all_parameters.csv" -outputPath "C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf"
```

Here is the output:
```
Using asm path: C:\Users\ngenesio\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml
Using yaml path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_all_options.yaml
Using csv path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\ERGOCUB_joint_all_parameters.csv
Using output path: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf
Conversion completed successfully. The URDF file is located at: C:\Users\ngenesio\icub-tech-iit\cad-libraries\output_urdf\model.urdf
```

> [!warning]
> This script has to be run in the `cad-libraries` folder to load the protk.dat

Until this will not become stable, this post will be the only documentation for using the plugin (ccing some people may use it cc @FabioBergonti @pillai-s @traversaro @xela-95 @mfussi66)

PR #86 (2024-04-05): Add the support for running the creo2urdf headless
I still have to manage to correctly pass the arguments to the plugin, and run it from terminal, but this should be required changes to make it possible.

It fixes #53 

cc @pattacini
closed 2024-04-05 by Nicogene