mesh-iit/creo2urdf #58: CREO2URDF – Write up documentation of classes

Nicogene, 2023-10-30:
We might add the documentation of the classes using doxygen and deploy it in a dedicated website

mfussi66, 2024-01-08:
I started this activity in the branch https://github.com/icub-tech-iit/creo2urdf/tree/doxygen , actually starting off from `Utils.h`, so that everything can be covered.

mfussi66, 2024-01-17:
I completed the documentation of the files, and added a Doxyfile to the branch. Additionally, I started setting up the github action, but first I need to study how it works.

Nicogene, 2024-01-17:
> I completed the documentation of the files, and added a Doxyfile to the branch. Additionally, I started setting up the github action, but first I need to study how it works.

You can take inspiration from
- ergocub-software: https://github.com/icub-tech-iit/ergocub-software/blob/master/.github/workflows/gh-pages.yml
- robometry: https://github.com/robotology/robometry/blob/master/.github/workflows/gh-pages.yml

They do the same thing with two different approaches

closed 2024-01-17 by Nicogene

Nicogene, 2024-01-17:
We are online:
- https://icub-tech-iit.github.io/creo2urdf/

PR #75 (2024-01-17): Add Doxygen documentation and gh-pages action
This PR adds:
- Doxygen documentation -> @mfussi66 
- The publish of the website created w/ mkdocs on gh-pages -> @Nicogene

It fixes #58