mesh-iit/creo2urdf #77: CREO2URDF – Wrong transform when exporting a sensors that uses a csys used already by a ft sensor

Nicogene, 2024-02-12:
Given a configuration file as follow:
```yaml
forceTorqueSensors: 
   - jointName: l_foot_front_ft_sensor
      directionChildToParent: No
      exportFrameInURDF: Yes
      frame: sensor
      linkName: l_ankle_2
      frameName: SCSYS_L_ANKLE_2_FT_FRONT
      sensorName: l_foot_front_ft
      sensorBlobs:
      - |
          <plugin name="left_foot_front_ft_plugin" filename="libgazebo_yarp_forcetorque.so">
            <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_foot_front_ft.ini</yarpConfigurationFile>
          </plugin>

sensors:
  - frameName: SCSYS_L_ANKLE_2_FT_FRONT
    linkName: l_ankle_2
    exportFrameInURDF: No # This is not since the frame is already exported in the forceTorqueSensors
    sensorName: l_foot_front_ft_imu
    sensorType: "accelerometer"
    updateRate: "100"
    sensorBlobs:
    - |
        <plugin name="ergocub_yarp_gazebo_plugin_IMU" filename="libgazebo_yarp_imu.so">
            <yarpConfigurationFile>model://ergoCub/conf/gazebo_ergocub_l_foot_front_ft_sensor_inertial.ini</yarpConfigurationFile>
        </plugin>
```

The ft sensor is correctly exported, on the other hand, the sensor (imu) is exported coincident w/ `linkName` frame 

``` xml
  <gazebo reference="l_ankle_2">
    <sensor name="l_foot_front_ft_imu" type="imu">
      <always_on>1</always_on>
      <update_rate>100.000000</update_rate>
      <pose>0 0 0 0 -0 0 </pose>
      <plugin name="ergocub_yarp_gazebo_plugin_IMU" filename="libgazebo_yarp_imu.so">
        <yarpConfigurationFile>model://ergoCub/conf/gazebo_ergocub_l_foot_front_ft_sensor_inertial.ini</yarpConfigurationFile>
      </plugin>
    </sensor>
  </gazebo>
  <sensor name="l_foot_front_ft_imu" type="accelerometer">
    <parent link="l_ankle_2"/>
    <origin rpy="0 -0 0 " xyz="0 0 0"/>
  </sensor>
  <gazebo reference="l_ankle_2">
    <sensor name="l_foot_rear_ft_imu" type="imu">
      <always_on>1</always_on>
      <update_rate>100.000000</update_rate>
      <pose>0 0 0 0 -0 0 </pose>
      <plugin name="ergocub_yarp_gazebo_plugin_IMU" filename="libgazebo_yarp_imu.so">
        <yarpConfigurationFile>model://ergoCub/conf/gazebo_ergocub_l_foot_rear_ft_sensor_inertial.ini</yarpConfigurationFile>
      </plugin>
    </sensor>
  </gazebo>
```

But instead there is a transform between the `frameName` and `linkName` cys:

[image]

cc @GiulioRomualdi @mfussi66

Nicogene, 2024-02-15:
Actually seems that it is not due to the fact that the frame is already used, because I tried to export an imu that uses the FT frame and I am getting the same (wrong) result indipendently both when I export also the ft and when I do not export it.

In the message window I am getting this error that is quite strange, since this frame is used for exporting the ft sensor, I will investigate further

```
l_foot_rear_imu: SCSYS_L_ANKLE_2_FT_REAR was not found
```

cc @mfussi66

Nicogene, 2024-02-15:
This problem seems to be related to:
- https://github.com/icub-tech-iit/creo2urdf/issues/60

With the solution proposed this issue should not happen if `ExportAllUserAdded` is set to `yes` but in our case we do not want to export all the frames, but still we want to be able to compute the correct transform for defining the sensor.

In particular, for the ft sensor, we do not use the `exportedFrames`:

https://github.com/icub-tech-iit/creo2urdf/blob/3d72a4507e43e3534495966f7e408a6e8b6dd252/src/creo2urdf/src/Sensorizer.cpp#L105-L125

On the other hand for the sensors we use the `exportedFrames`:

https://github.com/icub-tech-iit/creo2urdf/blob/3d72a4507e43e3534495966f7e408a6e8b6dd252/src/creo2urdf/src/Sensorizer.cpp#L221-L235

It should not to be too difficult to fix it, but I need to align w/ @mfussi66 for understanding why there is this difference

GiulioRomualdi, 2024-02-15:
It should be placed in the FT. Nothing prevents an error in the proposed xml blob. Can you generate the robot model so I can do a similar test done in https://github.com/icub-tech-iit/ergocub-software/pull/217#issuecomment-194137951 ?

Nicogene, 2024-02-16:
I think that the correct one seems to be the one I obtained after the fix, the ft sensor is not associated to a link but a joint, and the reference frame of a joint is the child link frame, looking these snippets

``` urdf
  <joint name="l_foot_rear_ft_sensor" type="fixed">
    <origin xyz="0.015999999999999986 -4.163336342344337e-17 -0.04789999999999994" rpy="-1.0970096963229695e-30 1.942890293093975e-16 2.0306083615826833e-17"/>
    <parent link="l_ankle_2"/>
    <child link="l_foot_rear"/>
  </joint>
```

``` urdf
  <gazebo reference="l_foot_rear_ft_sensor">
    <sensor name="l_foot_rear_ft" type="force_torque">
      <always_on>1</always_on>
      <update_rate>100</update_rate>
      <force_torque>
        <frame>sensor</frame>
        <measure_direction>parent_to_child</measure_direction>
      </force_torque>
      <pose>0 0 0 1.25667e-16 2.20112e-17 -2.0944 </pose>
      <plugin name="left_foot_rear_ft_plugin" filename="libgazebo_yarp_forcetorque.so">
        <yarpConfigurationFile>model://ergoCub/conf/FT/gazebo_ergocub_left_foot_rear_ft.ini</yarpConfigurationFile>
      </plugin>
    </sensor>
  </gazebo>
```

If we refer to `l_ankle_2` we should have the Z translation `-0.0479`. (Maybe `linkName` should be `l_foot_rear`)?

I don't know if it is sufficient but I have an urdf with only that imu added:
[model.zip](https://github.com/icub-tech-iit/creo2urdf/files/14308990/model.zip)

GiulioRomualdi, 2024-02-17:
Hi @Nicogene you are right!

I visualized your model and what I proposed in ergocub-software

### Your model
https://github.com/icub-tech-iit/creo2urdf/issues/77#issuecomment-1948030325

Here I asked to plot the following frames
```matlab
testFrames = {'l_foot_rear_imu', 'l_foot_rear_ft', 'l_sole'}; %% The frames to display
```

As you can notice `l_foot_rear_imu` and `l_foot_rear_ft` are coincident! 

[image]
 

### model I proposed in ergocub-software 

I asked to plot the same frame.

Here you can notice that the frame are not coincident! 

[image]

Nicogene, 2024-02-21:
Hi @GiulioRomualdi, if you or Guglielmo can double check these changes:

- https://github.com/icub-tech-iit/ergocub-software/pull/222

We can merge both the PR in ergocub-software and creo2urdf

commit efbf65b3 (2024-02-21): Fix the way we retrieve the transform for the sensors

This allows to export a sensor with a frame not specified in exportedFrames. It fixes #77.
commit cbf57b55 (2024-02-21): Fix the way we retrieve the transform for the sensors

This allows to export a sensor with a frame not specified in exportedFrames. It fixes #77.
commit 52274838 (2024-02-21): Fix the way we retrieve the transform for the sensors

This allows to export a sensor with a frame not specified in exportedFrames. It fixes #77.
commit e4419a69 (2024-02-29): Fix the way we retrieve the transform for the sensors

This allows to export a sensor with a frame not specified in exportedFrames. It fixes #77.
closed 2024-02-29 by Nicogene