mesh-iit/creo2urdf #84: CREO2URDF – The dll pre-compiled seems to not work on pc different from the one who compiled it

Nicogene, 2024-03-12:
As reported in:
- https://github.com/orgs/robotology/discussions/688

And by @divyashah, it seems that it is not possible to use the dll we provide in the releases using a pc different from the one created it.

It happened also to @mfussi66, it seems something related to the license.

cc @pattacini @Agrobotdev

Nicogene, 2024-03-14:
@mfussi66 opened a ticket:
- https://community.ptc.com/t5/Customization/EXCEPTION-NONCONTINUABLE-EXCEPTION-when-using-OTK-dll-compiled/m-p/936942

We may need to unlock the dll, and digitally sign it(?)

cc @maggia80 @pattacini

maggia80, 2024-03-14:
In case we need to purchase a license we can do that

Nicogene, 2024-03-19:
Today I tried to debug it with the help of @mfussi66 and it not seems a problem of being compiled by another user but rather seems to be related to the version of the VS installed and the version of the VS used for compiling the dll.

It happens that it crashed also a dll that I compiled while ago

Nicogene, 2024-03-19:
At the end, the reason of the crash during the execution was not due to some license/dependencies issue, but it was that the `v0.4.5` dll did not contain these changes

- https://github.com/icub-tech-iit/creo2urdf/pull/76

That are required for running the simple `2bars` example.

I re-tagged the latest master that is working fine uploading the .zip

- https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.4.6

It worked both for @mfussi66, @agrobotdev you can give a try with the latest version released.

On the other hand @divyashah is experiencing a problem on the startup/load of the plugin, not on the execution, so it may be a different problem

Nicogene, 2024-03-20:
Another interesting piece of information, @salvi-mattia managed to correctly install and use the `creo2urdf` loading the dll in the latest release so it may be that @divyashah may have something corrupted in the creo installation, looking at PTC Creo Community  for the `EXCEPTION_NONCONTINUABLE_EXCEPTION` they advise to reinstall it, see:

- https://community.ptc.com/t5/3D-Part-Assembly-Design/EXCEPTION-NONCONTINUABLE-EXCEPTION/td-p/765266

Nicogene, 2024-04-09:
I managed to install(via binaries) and run creo2urdf also on our shared machine that does not have VS installed:
- https://github.com/icub-tech-iit/creo2urdf/issues/87#issuecomment-2044679835

I think we can close it and assist @divyashah offline.