mesh-iit/creo2urdf #85: Unable to get the transform from to the root for AA

JunzheJosephZhu, 2024-03-29:
[image]
I'm using Creo 9.0.2, and this is the setup. I'm using a very simple dummy ASM with only two links and one joint.
Could anyone identify the issue for me please? Thanks in advance!

mfussi66, 2024-03-29:
Hi @JunzheJosephZhu , thanks for using the extension and providing feedback! Is the warning you posted the only one? Can you expand the message window to check for other warnings?

Also, it's a good chance to improve the warning message.

JunzheJosephZhu, 2024-03-29:
Hi,
[image]
Below is the screenshot of the full error message

traversaro, 2024-03-29:
@JunzheJosephZhu can you share with us the dummy asm you are testing? Thanks!

JunzheJosephZhu, 2024-03-29:
[A.zip](https://github.com/icub-tech-iit/creo2urdf/files/14805154/A.zip)
Hi, This zip file contains the asm file as well as the prt files and the configuration files

mfussi66, 2024-03-30:
Hi @JunzheJosephZhu, I looked into the assembly, and I found two things:
- The coordinate systems that belong to the parts have different names from the default `CSYS`:
    - if you'd like to use those names, you need to specify the names in the `linkFrames` section of the .yaml file
    - Or you can rename them to `CSYS`, BUT, and here goes the second thing:
- Creo crashes if I set them to `CSYS`. This is because the plugin tries to build a joint between CC and BB, and not AA and BB. I think it is because, to define the joint, you are using for AA a protrusion axis and not a dedicated axis datum, unlike BB which uses the axis A_1. Therefore, my suggestion is to create a datum axis for AA and try again. This is the way we build _urdf-iable_ assemblies internally

**In detail:**
The problem lies here(https://github.com/icub-tech-iit/creo2urdf/blob/master/src/creo2urdf/src/ElementTreeManager.cpp#L189C1-L196C14). Due to the obtusity of the Creo Toolkit, this was the only way I found to get the name of two parts that create the coincident axes constraint. For some reason, on the last iteration we get CC and BB, and not AA and BB.

Therefore, when `link_info_map.at(parent)` is called, it looks for the key "CC", which does not exist, and it crashes after throwing an unhandled exception.