mesh-iit/creo2urdf #87: CREO2URDF – Build a GA on a self-hosted runner for automatically generate the urdfs

Nicogene, 2024-04-05:
### Task description

Since now `creo2urdf` can be run from terminal:
- https://github.com/icub-tech-iit/creo2urdf/issues/53

and we have `run_creo2urdf.ps1` script, we need to prepare our self-hosted runner for generating automatically the `ergocub-software models`

### DoD

The github action and the self-hosted runner are ready.

pattacini, 2024-04-06:
Because of https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/about-self-hosted-runners#self-hosted-runner-security, we ought to rely on a private repository.

I've churned out a brand-new one:
- https://github.com/icub-tech-iit/creo2urdf-private

That said, it could be still beneficial for the community to document here (i.e., public-wise) how to set up the infrastructure for the GA. 

That being said, @Nicogene, could you please draft below what the GitHub Action is supposed to be doing in terms of input/output? This will help me to organize the work.

Nicogene, 2024-04-08:
As agreed f2f w/ @pattacini, we thought about creating a central GitHub action in the `creo2urdf-private` repo that handles the generation of urdf coming hosted on several repositories (e.g. `ergocub-software`, `icub-models-generator`).
This GA will run on our self-hosted runner, the idea is that a commit on the master branch of a repository triggers a `repository dispatch` that will be received by this new GitHub action that will generate the URDFs and will open a PR on that repository.

ccing @traversaro since he may be interested

traversaro, 2024-04-08:
If I am not wrong, we should have trigger repository on both the repos that host the configuration files and the one that host the CAD, right?

pattacini, 2024-04-08:
Ideally yes, @traversaro.

To start with, I would honestly send the triggers via the green button (i.e., `workflow_dispacth`).
This will allow us to check all the intermediate steps on both ways.

Nicogene, 2024-04-09:
I managed to install and run `creo2urdf` in our shared windows machine. I put all the necessary repositories/files under `Users/Public/icub-tech-iit` in this way they should be accessible by any logged user, and I put as creo starting folder `Public/icub-tech-iit/cad-libraries`.

I will move on setting the self hosted runner

cc @davidelasagna @pattacini @mfussi66

Nicogene, 2024-04-09:
I built a first draft of the GA that runs from a manual dispatch that simply execute this commands:
```ps
          cd C:\Users\Public\icub-tech-iit\cad-libraries
          mkdir output_urdf
          .\run_creo2urdf.ps1 -asmPath "C:\Users\Public\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm" -yamlPath "$env:GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_all_options.yaml" -csvPath "$env:GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_joint_all_parameters.csv" -outputPath "C:\Users\Public\icub-tech-iit\cad-libraries\output_urdf"
          cp .\output_urdf\model.urdf $env:GITHUB_WORKSPACE\ergocub-software\urdf\ergoCub\robots\ergoCubSN001
          .\run_creo2urdf.ps1 -asmPath "C:\Users\Public\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm" -yamlPath "$env:GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_all_options_gazebo.yaml" -csvPath "$GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_joint_all_parameters.csv" -outputPath "C:\Users\Public\icub-tech-iit\cad-libraries\output_urdf"
          cp .\output_urdf\model.urdf $env:GITHUB_WORKSPACE\ergocub-software\urdf\ergoCub\robots\ergoCubGazeboV1_1
          .\run_creo2urdf.ps1 -asmPath "C:\Users\Public\icub-tech-iit\cad-mechanics\projects\simulation_model\ergocub_1-1\sim_ergocub_1-1.asm" -yamlPath "$env:GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_all_options_minContacts.yaml" -csvPath "$GITHUB_WORKSPACE\ergocub-software\urdf\creo2urdf\data\ergocub1_1\ERGOCUB_joint_all_parameters.csv" -outputPath "C:\Users\Public\icub-tech-iit\cad-libraries\output_urdf"
          cp .\output_urdf\model.urdf $env:GITHUB_WORKSPACE\ergocub-software\urdf\ergoCub\robots\ergoCubGazeboV1_1_minContacts
          cp .\output_urdf\*.stl $env:GITHUB_WORKSPACE\ergocub-software\urdf\ergoCub\meshes\simmechanics
          rmdir -Recurse C:\Users\Public\icub-tech-iit\cad-libraries\output_urdf
```

The problem seems that the first time the urdf is generated fine, then creo seems to remain zombie and the following generations fails, I need to investigate a little bit about it.

Nicogene, 2024-04-10:
The strange thing is that I can run `creo2urdf` from the shared machine using the GUI, but it does not work via the command line starts and immediately stops.
In theory, I am using the very same version of `creo2urdf`, `cad-libraries`, `cad-mechanics`, the only differernce is that when I load the simulation model asm I am getting this error on the shared machine:
```
Could not read from Model Tree config file .\config\tree.cfg.
```

`.\config\tree.cfg` is not present in my cad-libraries but for some reason, I am not getting that error.

@Lawproto @fiorisi do you know what that file is for? Is it necessary?

Nicogene, 2024-04-10:
> In theory, I am using the very same version of creo2urdf, cad-libraries, cad-mechanics, the only differernce is that when I load the simulation model asm I am getting this error on the shared machine:

Actually now I pulled `cad-libraries` and `cad-mechanics` and the problem is happening also on my machine 👀

Nicogene, 2024-04-10:
Bingo! Here what was the problem:

> 'sim_ergocub_1-1' cannot be retrieved.

I remembered I had these local changes in cad-libraries:
```diff
--- a/config/search_path.pro
+++ b/config/search_path.pro
@@ -85,6 +85,8 @@ search_path ".\..\cad-mechanics\projects\icub\koroibot"
 search_path ".\..\cad-mechanics\projects\simulation_model"
 search_path ".\..\cad-mechanics\projects\simulation_model\head"
 search_path ".\..\cad-mechanics\projects\simulation_model\icub3"
+search_path ".\..\cad-mechanics\projects\simulation_model\ergocub_1-1"
+search_path ".\..\cad-mechanics\projects\simulation_model\cer"
```

cc @fiorisi

pattacini, 2024-04-10:
>Shall we clone cad-libraries, cad-mechanics from scratch ? (they are quite heavy, and we should repeat the installation of creo2urdf)

To start with, we could just guarantee that those repos are aligned via a pull with what is needed to generate the URDF.

Nicogene, 2024-04-10:
Fix in `cad-libraries`:
- https://github.com/icub-tech-iit/cad-libraries/pull/178

Nicogene, 2024-04-11:
Unfortunately sometimes some creo background process remains stuck even if the generation exit fine, I put a wait of 30 sec with a forced kill of the process

- https://github.com/icub-tech-iit/creo2urdf/commit/dabaeaa3eaa5954be30e5ace0763aef6203cd3e6

Nicogene, 2024-04-11:
With the latest version of the action the output of the generation is a PR in `ergocub-software`:
- https://github.com/icub-tech-iit/ergocub-software/pull/237

@xela-95 @traversaro now the versions of all the repo used is both in the commit message and PR body

Nicogene, 2024-04-12:
An MVP has been achieved, closing in favour of:

- https://github.com/icub-tech-iit/creo2urdf/issues/90

closed 2024-04-12 by Nicogene