mesh-iit/creo2urdf #100: Impossible to generate URDF of three links model with "Element [...] is not found in configuration file"

DavideBarbone, 2024-07-08:
Hello, thanks a lot for the work on this repo. I am trying to export the URDF of a three link Creo model, starting from the 2bars example. The 2bars example is working fine, but when I try to export my model (that I can share inside IIT) with this configuration file:

~~~
robotName: sim_trailer
scale: [0.001,0.001,0.001]
root: FRAME_SW0001
~~~

and .csv:

~~~
joint_name,lower_limit,upper_limit,damping,friction,velocity_limit,effort_limit
~~~

The export creates an error:

[image]

and then Creo crashes. I am not sure why this is happening, as the 2bars example is working fine on my machine (I am using Creo 9.0.4.0). However, investigating in the code I noticed that the errors seems related to the `rename` option, and so by modifying the file adding:

~~~
rename:
    SIM_TRAILER--FRAME_SW0001: base_link1
~~~

I get a different error:

[image]

DavideBarbone, 2024-07-08:
fyi @Nicogene @mfussi66 @traversaro

Nicogene, 2024-07-08:
Hi @DavideBarbone!

Thanks for reporting, this behaviour is strange because the conf files used in the 2bars are similar to the ones you are using, which version of `creo2urdf` are you using?

Could you please post the `traceback,log` ? It should be in the creo working directory, it is created when it crashes.

traversaro, 2024-07-08:
> (The warnings should be fine)

Kind of OT: Looking at the code in https://github.com/icub-tech-iit/creo2urdf/blob/8ee8549a2d00f65a05385e79ffad2edb6fac2ef4/src/creo2urdf/src/Creo2Urdf.cpp#L821 indeed it seems that the warning is printed if there is no `rename` is provided, the warning is printed. However, the docs does not mention at all that rename should be provided for all elements. Probably if this is not a problem we should not print a warning?

DavideBarbone, 2024-07-08:
Hi @Nicogene!

Thanks for the help, I downloaded creo2urdf from https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.5.0.

The traceback.log file is the following:

``` -------------------------------------------------
The trace creation timestamp is: Mon Jul 08 16:27:20 24
The executable build timestamp is: Thu Mar 16 16:40:58 23
The datecode is: 2023014
The pro machine type is: x86e_win64
The process ID is: 13016
The traceback type is : CRASH
The logger exit status is: 01003
The Trail is: trail.txt.36
The release is: 9.0
The product version is: 9.0.4.0
The product title is: Creo Parametric
-------------------------------------------------
Exception EXCEPTION_NONCONTINUABLE_EXCEPTION has occurred in the thread 18164.

Main Thread 18164
=====================
0x00007FFBE471F77D	NULL (NULL:0) (ntdll:0x00007FFBE4610000) ( 0x735ac840, 0xb71f0000, 0xc0102b54, 0xc8d17aac ) 
0x00007FFBE46C7ABB	NULL (NULL:0) (ntdll:0x00007FFBE4610000) ( 0x0, 0x735ac6f0, 0x0, 0xe46643c5 ) 
0x00007FFBE4664465	NULL (NULL:0) (ntdll:0x00007FFBE4610000) ( 0x0, 0x7dfa6708, 0x735ac980, 0xadf587f0 ) 
0x00007FFBE199543C	NULL (NULL:0) (KERNELBASE:0x00007FFBE1930000) ( 0xe10ff560, 0xe464ab11, 0xb1d6dac0, 0x7df79a5c ) 
0x00007FFBB1FA6220	NULL (NULL:0) (VCRUNTIME140:0x00007FFBB1FA0000) ( 0x10b249ab, 0x3e445f8, 0xe101ec30, 0xb1d6dac0 ) 
0x00007FFB7DF79DC2	NULL (NULL:0) (MSVCP140:0x00007FFB7DF40000) ( 0xe101ec30, 0x1e00000, 0x735acd00, 0xe101ec30 ) 
0x00007FFB03E3C9C9	NULL (NULL:0) (creo2urdf:0x00007FFB03D60000) ( 0xadf58910, 0xe1059670, 0x0, 0xadf58910 ) 
0x00007FFB0479118C	NULL (NULL:0) (creo2urdf:0x00007FFB03D60000) ( 0xcd7242b0, 0x0, 0x0, 0xadf58910 ) 
0x00007FFB39F6BDD8	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0x0, 0x2048fe, 0x800000, 0x800004 ) 
0x00007FFB39E7A3E5	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0x0, 0x0, 0x800004, 0xaa2b81b0 ) 
0x00007FFBAA2BE496	NULL (NULL:0) (libdui_sh:0x00007FFBAA2B0000) ( 0x311066c, 0x0, 0x0, 0x735adc30 ) 
0x00007FFBAA2B81B9	NULL (NULL:0) (libdui_sh:0x00007FFBAA2B0000) ( 0x735adc30, 0x735ade50, 0x0, 0xffffffff ) 
0x00007FF6BE56C242	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FF6BE56F9D4	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x23, 0x735ade50, 0x735adda0, 0xffffffff ) 
0x00007FF6BE56C8C4	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x1, 0x0, 0x23, 0x0 ) 
0x00007FF6BE578127	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x1, 0xffffffff, 0x735ae0f0, 0xc3d22438 ) 
0x00007FF6BE571724	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFB4D7FB7FB	NULL (NULL:0) (coretools_sh:0x00007FFB4D7B0000) ( 0x0, 0x80000d, 0x80000d, 0x0 ) 
0x00007FF6BE577ED6	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x0, 0x0, 0xcd88c588 ) 
0x00007FF6B84253FF	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x0, 0x80000d, 0x80000d ) 
0x00007FF6B927AADF	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x1, 0x0, 0x0 ) 
0x00007FF6B9287E38	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xc1ec43c8, 0x0, 0x0, 0x0 ) 
0x00007FF6B9287CC9	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xc1ec43c8, 0x735aee30, 0x1, 0x0 ) 
0x00007FF6B7FDDFF7	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x4, 0x735ae570, 0x0 ) 
0x00007FF6B800078D	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x735aea88, 0xffffffff, 0xaa4283af ) 
0x00007FF6B80229F6	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xb7f06c00, 0x0, 0x6, 0x800005 ) 
0x00007FF6B8022C77	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xc1ec43c8, 0x735aeab0, 0x800005, 0xcd7422f8 ) 
0x00007FF6B7F06B5D	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x1, 0xffffffff, 0x0, 0xffffffff ) 
0x00007FF6B7F052C9	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xc1ec43c8, 0xc76eb790, 0xffffffff, 0x800005 ) 
0x00007FF6B7FEDDFD	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xbebe8a68, 0xffffffff, 0x39fdd880, 0xbc8eec48 ) 
0x00007FFB39F6BDD8	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0x0, 0x2048fe, 0x800000, 0x800005 ) 
0x00007FFB39E7A3E5	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0x0, 0x1, 0x800005, 0xaa2b81b0 ) 
0x00007FFBAA2BE496	NULL (NULL:0) (libdui_sh:0x00007FFBAA2B0000) ( 0x80000001, 0x0, 0x0, 0x735af350 ) 
0x00007FFBAA2B81B9	NULL (NULL:0) (libdui_sh:0x00007FFBAA2B0000) ( 0x735af350, 0x735af3a0, 0xbc800410, 0xffffffff ) 
0x00007FF6BE56C242	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0xbc800410, 0xbc800458, 0x0 ) 
0x00007FF6BE56C96E	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x0, 0x0, 0x24, 0x0 ) 
0x00007FF6BE578127	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x1, 0xffffffff, 0x735af640, 0xc3d22438 ) 
0x00007FF6BE571724	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x130b7982, 0x1ce19183, 0x2271d4d3, 0x2ec80928 ) 
0x00007FFB4D7FB7FB	NULL (NULL:0) (coretools_sh:0x00007FFB4D7B0000) ( 0xc3732270, 0x3a0578b0, 0x3, 0x52b27efe ) 
0x00007FF6BE577ED6	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x3a0578b0, 0x0, 0x0, 0xe1da16e0 ) 
0x00007FF6B8022EE6	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xb0748c68, 0x3, 0x0, 0x0 ) 
0x00007FF6B7F21303	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xb0748c68, 0x0, 0x3a0578b0, 0x3 ) 
0x00007FF6C01028A5	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0x1, 0x5a9b6a4, 0x42eb18f0, 0x43b2a504 ) 
0x00007FF6C0102B54	NULL (NULL:0) (xtop:0x00007FF6B71F0000) ( 0xb0748c68, 0xc01028d0, 0xae443b60, 0x735afb70 ) 
0x00007FFB39F1A57C	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0xb0748c68, 0xc01028d0, 0x735afb70, 0x0 ) 
0x00007FFB39F1B60F	NULL (NULL:0) (libui_sh:0x00007FFB39CF0000) ( 0xc01028d0, 0x0, 0x0, 0xc3732190 ) 
[... cut ...]
0x00007FFBC61E230B	NULL (NULL:0) (clr:0x00007FFBC6170000) ( 0xe103f2c0, 0xffffffff, 0x740f248, 0x740f2d0 ) 
0x00007FFBC61E222F	NULL (NULL:0) (clr:0x00007FFBC6170000) ( 0xe103f2c0, 0x0, 0x740f248, 0x740f2d0 ) 
0x00007FFBC61E20FB	NULL (NULL:0) (clr:0x00007FFBC6170000) ( 0xffffffff, 0xe103f2c0, 0x0, 0x1 ) 
0x00007FFBC627A2A7	NULL (NULL:0) (clr:0x00007FFBC6170000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFBC62E58EA	NULL (NULL:0) (clr:0x00007FFBC6170000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFBE325257D	NULL (NULL:0) (KERNEL32:0x00007FFBE3240000) ( 0x0, 0x0, 0x0, 0x0 ) 
```

Nicogene, 2024-07-08:
> @Nicogene if useful @DavideBarbone can share the model to you (we can't share it here as it is from a commercial project). 

Yes sure!

The strange thing is that that traceback does not contain any information, and `EXCEPTION_NONCONTINUABLE_EXCEPTION` that is a problem that @mfussi66 and I encountered in the past but never understood what can throw the exception.

Nicogene, 2024-07-09:
Today I tried to export the urdf and Creo is not crashing, I am just getting this error from idyntree:
```
[ERROR] Model :: addLink : a link or frame of name RUOTA_SW0001 is already present in the model
[ERROR] Model :: getLinkIndex : Impossible to find link FRAME_SW0001 in the Model
[ERROR] URDFStringFromModel : specified baseLink FRAME_SW0001 is not part of the model
```

Probably the creo2urdf version I am using (latest master) contains some fixes that allow to gently fail instead of crash when 2 links have the same name.

DavideBarbone, 2024-07-09:
Hi @Nicogene, I renamed the two parts, the plugin still doesn't work, and creo crashes with the same mode, but now the error is different:

[image]

The yaml file that I am using is the following:

```
robotName: sim_trailer
scale: [0.001,0.001,0.001]
root: FRAME_SW0001

rename:
    SIM_TRAILER--FRAME_SW0001: base_link1
    FRAME_SW0001: base_link
    FRAME_SW0001--L_WHEEL: left_wheel1
    L_WHEEL: left_wheel
    FRAME_SW0001--R_WHEEL: right_wheel1
    R_WHEEL: right_wheel
```

<details>
  <summary>Traceback file</summary>
 
```-------------------------------------------------
The trace creation timestamp is: Tue Jul 09 15:14:17 24
The executable build timestamp is: Thu Mar 16 16:40:58 23
The datecode is: 2023014
The pro machine type is: x86e_win64
The process ID is: 16256
The traceback type is : CRASH
The logger exit status is: 01003
The Trail is: trail.txt.42
The release is: 9.0
The product version is: 9.0.4.0
The product title is: Creo Parametric
-------------------------------------------------
Exception EXCEPTION_NONCONTINUABLE_EXCEPTION has occurred in the thread 11268.

Main Thread 11268
=====================
0x00007FFD912DF77D	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0xa41ac920, 0x30690000, 0x395a2b54, 0x421b7aac ) 
0x00007FFD91287ABB	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0x0, 0xa41ac7d0, 0x0, 0x912243c5 ) 
0x00007FFD91224465	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0x0, 0x48616708, 0xa41aca60, 0xe8bb9ed0 ) 
0x00007FFD8E5D543C	NULL (NULL:0) (KERNELBASE:0x00007FFD8E570000) ( 0xf1ceef60, 0x9120ab11, 0xda134570, 0x485e9a5c ) 
0x00007FFD68096220	NULL (NULL:0) (VCRUNTIME140:0x00007FFD68090000) ( 0x785d9a5c, 0xcaba45f8, 0x94f8c290, 0xda134570 ) 
0x00007FFD485E9DC2	NULL (NULL:0) (MSVCP140:0x00007FFD485B0000) ( 0x94f8c290, 0x1e00000, 0xa41ace00, 0x94f8c290 ) 
0x00007FFCCAB9C9C9	NULL (NULL:0) (creo2urdf:0x00007FFCCAAC0000) ( 0xe8bb9ff0, 0xf1845550, 0x0, 0xe8bb9ff0 ) 
0x00007FFCCB4F118C	NULL (NULL:0) (creo2urdf:0x00007FFCCAAC0000) ( 0x80364558, 0x0, 0x0, 0xe8bb9ff0 ) 
0x00007FFCE564BDD8	NULL (NULL:0) (libui_sh:0x00007FFCE53D0000) ( 0x0, 0x2048fe, 0x800000, 0x800004 ) 
0x00007FFCE555A3E5	NULL (NULL:0) (libui_sh:0x00007FFCE53D0000) ( 0x0, 0x0, 0x800004, 0x63d981b0 ) 
0x00007FFD63D9E496	NULL (NULL:0) (libdui_sh:0x00007FFD63D90000) ( 0x1510969, 0x0, 0x0, 0xa41add10 ) 
0x00007FFD63D981B9	NULL (NULL:0) (libdui_sh:0x00007FFD63D90000) ( 0xa41add10, 0xa41adf30, 0x0, 0xffffffff ) 
0x00007FF737A0C242	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FF737A0F9D4	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x23, 0xa41adf30, 0xa41ade80, 0xffffffff ) 
0x00007FF737A0C8C4	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x1, 0x0, 0x23, 0x0 ) 
0x00007FF737A18127	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x1, 0xffffffff, 0xa41ae1d0, 0x3d1c2438 ) 
0x00007FF737A11724	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD0CA0B7FB	NULL (NULL:0) (coretools_sh:0x00007FFD0C9C0000) ( 0x0, 0x80000d, 0x80000d, 0x0 ) 
0x00007FF737A17ED6	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x0, 0x0, 0x8039c648 ) 
0x00007FF7318C53FF	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x0, 0x80000d, 0x80000d ) 
0x00007FF73271AADF	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x1, 0x0, 0x0 ) 
0x00007FF732727E38	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0xdfae4468, 0x0, 0x0, 0x0 ) 
0x00007FF732727CC9	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0xdfae4468, 0xa41aef10, 0x1, 0x0 ) 
0x00007FF73147DFF7	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0x4, 0xa41ae650, 0x0 ) 
0x00007FF7314A078D	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x0, 0xa41aeb68, 0xffffffff, 0x5d7583af ) 
0x00007FF7314C29F6	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x313a6c00, 0x0, 0x6, 0x800005 ) 
0x00007FF7314C2C77	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0xdfae4468, 0xa41aeb90, 0x800005, 0x80c60bd8 ) 
0x00007FF7313A6B5D	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0x1, 0xffffffff, 0x0, 0xffffffff ) 
0x00007FF7313A52C9	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0xdfae4468, 0x40b8b790, 0xffffffff, 0x800005 ) 
0x00007FF73148DDFD	NULL (NULL:0) (xtop:0x00007FF730690000) ( 0xe9ca8a68, 0xffffffff, 0xe56bd880, 0xe395ec48 ) 
[... 428 lines cut ...]
0x00007FFD9120537E	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD8F64257D	NULL (NULL:0) (KERNEL32:0x00007FFD8F630000) ( 0x0, 0x0, 0x0, 0x0 ) 

Thread 15132
=====================
0x00007FFD912704A4	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0x0, 0x750047, 0x500064, 0x0 ) 
0x00007FFD8E5CFDA9	NULL (NULL:0) (KERNELBASE:0x00007FFD8E570000) ( 0x94aff420, 0xf1da15c0, 0x94aff420, 0xf1da15c0 ) 
0x00007FFD71D078C6	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x94aff420, 0x94aff420, 0x94aff458, 0xf1da15c0 ) 
0x00007FFD71D077AC	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD71D07646	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD8F64257D	NULL (NULL:0) (KERNEL32:0x00007FFD8F630000) ( 0x0, 0x0, 0x0, 0x0 ) 

Thread 10516
=====================
0x00007FFD912704A4	NULL (NULL:0) (ntdll:0x00007FFD911D0000) ( 0xb848f450, 0x0, 0x1a2ac700, 0x71cab1ea ) 
0x00007FFD8E5CFDA9	NULL (NULL:0) (KERNELBASE:0x00007FFD8E570000) ( 0xb848fa30, 0x1, 0x0, 0x1 ) 
0x00007FFD71BD1D6F	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x0, 0x71d2e558, 0x0, 0xb848f8e0 ) 
0x00007FFD71BD1B20	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0xb848f918, 0xb848f918, 0xb848f3c8, 0xb848f450 ) 
0x00007FFD71C1230B	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x94bedaa0, 0xffffffff, 0xb848f3c8, 0xb848f450 ) 
0x00007FFD71C1222F	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x94bedaa0, 0x0, 0xb848f3c8, 0xb848f450 ) 
0x00007FFD71C120FB	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0xffffffff, 0x94bedaa0, 0x0, 0x1 ) 
0x00007FFD71CAA2A7	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD71D158EA	NULL (NULL:0) (clr:0x00007FFD71BA0000) ( 0x0, 0x0, 0x0, 0x0 ) 
0x00007FFD8F64257D	NULL (NULL:0) (KERNEL32:0x00007FFD8F630000) ( 0x0, 0x0, 0x0, 0x0 )```
</details>

Nicogene, 2024-07-09:
Probably your crash is due to a limitation that creo2urdf had, when different parts are using link frames with the same name (`CSYS` in your case), this should be fixed in : 
- #98 

I haven't produced a new release/binary yet but if you send me an updated version of the asm I can check it, and then I can release it.

Moreover, `creo2urdf` gives its best if the assembly used conforms with [our internal guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Prepare-PTC-Creo-Mechanism-for-URDF#additional-datums).
If you use a link frame a link that is not defined as follow:
```
    a link frame SCSYS_link_name defined as follow:
        X-axis pointing forward
        Z-axis pointing up
        Y-axis to complete the right-handed triplet
        one of the axes has to be coincident with the parent joint axis

when it's not applicable:

    Z-axis is the one pointing upwards at the most
    X-axis pointing forward at the most
    Y-axis to complete the right-handed triplet
    one of the axes has to be coincident with the parent joint axis

```
(Note that for each link not present in the `linkFrames` section of the yaml is not specified CSYS is used instead)

the resulting joints will probably not be defined as expected.
In our case, using `CSYS` as `linkFrame` is **NOT** ok, `SCSYS_WHEEL` is ok.

See:
- #99

Nicogene, 2024-07-12:
By using this new `creo2urdf` version:
- https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.5.1

And fixing the yaml used by @DavideBarbone, we managed to export the assembly correctly.

Idyntree failed to export the urdf because the `root` parameter was set to a link that then was renamed, setting to the renamed valued made export the urdf correctly.

Closing!

closed 2024-07-12 by Nicogene