mesh-iit/creo2urdf #101: Rename ribbon from "iCubTech" to creo2urdf ?

traversaro, 2024-07-08:
While working with @DavideBarbone, I realized it is a bit confusing for non-IIT users for creo2urdf to understand why they need to navigate to the "iCubTech" ribbon to actually find the "creo2urdf" button. Could it make sense to rename the "iCubTech" ribbon to creo2urdf ?

Nicogene, 2024-07-08:
You are right, maybe it is better to place it in the `Tools` ribbon as the `Mathworks` one

mfussi66, 2024-08-09:
I created a new ribbon file that addresses the issue. However, it seems impossible to "inject" the creo2urdf button in the Mathworks group, so I did the following:

[image]

Let me know if it's ok.

The validation button can be removed as it's not used at the moment.

traversaro, 2024-08-09:
I like it! The only minor comment is that URDF is an acronym, so in general it does not make a lot of sense to capitalize it, so I would either write URDF or urdf, while Urdf looks a bit strange (see for example how UDF Library is capitalized in the Utilities Menu).

mfussi66, 2024-08-09:
Coming right up:

[image]

Nicogene, 2024-08-23:
Fixed by #108

closed 2024-08-23 by Nicogene