mesh-iit/creo2urdf #102: Remove "Validate" button?

traversaro, 2024-07-08:
Working with @davidebarbone, I noticed it was a bit confusing to have a "Validate" button that does not work. If we are not using the button at the moment, could it make sense to just remove it?

Nicogene, 2024-07-08:
This was a button that worked for simple assemblies but the validation joint per joint is not feasible for complex assemblies, and for the time being we decided just to "disconnect" it, but if its presence is confusing we can remove it from the ribbon it should be trivial

traversaro, 2024-07-08:
What is the advantage of "disconnecting" it as opposed to just remove it?

Nicogene, 2024-07-09:
> What is the advantage of "disconnecting" it as opposed to just remove it? At least from the GUI, then the code can still be there, but at least it does not confuse new users.

Totally agree, it has probably been done to remove it quickly

Nicogene, 2024-08-23:
Fixed by #108

closed 2024-08-23 by Nicogene