mesh-iit/creo2urdf #112: Unhandled exception if a part contains a coordinat system with unexpected name

mfussi66, 2024-11-11:
There might be instances in which a Creo part does not contain a coordinate system called CSYS (its default name), or one with the name defined in the section `linkFrames`. 

When it happens, an unhandled exception occurs, the plugin quits silently and an attempt to relaunch it does not launch it cleanly. 

It is therefore necessary to handle this type of exception.

mfussi66, 2024-11-11:
Using the debugger, I could see that the code stops working when `comp_path` calls `GetTransform`, in the function `getTransformFromOwnerToLinkFrame`.

A first step to fix this issue is wrapping the problem:

```cpp
getTransformFromOwnerToLinkFrame(pfcComponentPath_ptr comp_path, 
    pfcModel_ptr modelhdl, 
    const std::string& link_frame_name, 
    const array<double, 3>& scale) {
    
    iDynTree::Transform csysAsm_H_link = iDynTree::Transform::Identity();
    auto csysAsm_H_csysPart = iDynTree::Transform::Identity();
    try {
        csysAsm_H_csysPart = fromCreo(comp_path->GetTransform(xtrue), scale);
    }
    xcatchbegin
    xcatchcip(defaultEx)
    {
        printToMessageWindow("Exception caught: Could not retrieve transform of " + link_frame_name, c2uLogLevel::WARN);
    }
    xcatchend
```

mfussi66, 2024-11-11:
However, the mass [is not a mandatory parameter](http://sdformat.org/spec?ver=1.11&elem=link#inertial_mass) for a link, so maybe thats not the right approach?

cc @traversaro

traversaro, 2024-11-11:
I am not sure I get the relation between the two aspects, perhaps we can discuss tomorrow in person?