mesh-iit/creo2urdf #113: Test creo2urdf with creo 11

Nicogene, 2024-11-11:
We should try `creo2urdf` with creo11 before the facility does the transition from 9->11.

cc @fiorisi @mfussi66

Nicogene, 2025-03-31:
I would like to address this first:
- #121

Nicogene, 2025-03-31:
Surprisingly, it just works w/ a creo2urdf compiled using the creo9 API:

[image]

I will try also w/ creo11 API

Nicogene, 2025-03-31:
It works fine also compiling w/ creo11 API.

I tested both ergoCub and iCub simulation model.

cc @pattacini @maggia80 @fiorisi

closed 2025-03-31 by Nicogene

Nicogene, 2025-04-02:
Interestingly since creo2urdf compiled w/ creo 9 API works w/ creo 11, the other way around does not work:

[image]

traversaro, 2025-04-02:
> Interestingly since creo2urdf compiled w/ creo 9 API works w/ creo 11, the other way around does not work:
> 
> [image]
> 
> We should keep in mind when delivering binaries in the future [@mfussi66](https://github.com/mfussi66)

Probably Creo mantains a stable ABI, so .dll compiler with an earlier version of Creo can work in later version, but not the other way around, as even if the ABI is stable, it is not immutable, in the sense that new symbols can be added.