mesh-iit/creo2urdf #126: It does not work with Creo education version

CyrusTZ, 2025-04-11:
Hi,

At first, I tried adding the "creo2urdf" from the procedures explained [here](https://github.com/icub-tech-iit/creo2urdf/tree/master?tab=readme-ov-file#creo2urdf) to Creo version 10.0.4.0 education, with no luck. I uninstalled version 10.0.4.0 and installed version 11.0.0.0, which is the only latest version available from the Creo PTC website for education, still no sign of success.

Could you please confirm if I am mistaken or if the extension does not work with the education versions?

Thanks in advance.

mfussi66, 2025-04-11:
What error are you getting? Does Creo fail silently? With more info we can check internally if there are licensing issues.

mfussi66, 2025-04-17:
Apologies for the late response.

> Could you please confirm if I am mistaken or if the extension does not work with the education versions?

We internally use the Education version as well, so hopefully it's not an issue. Did you make sure to install the c++ object toolkit ? 

Can you compile the plugin from sources?

CyrusTZ, 2025-04-17:
Hello Mattia,

Don't worry about it, and thanks.

-> Did you make sure to install the c++ object toolkit ?

It was needed for the[ installation from source](https://github.com/icub-tech-iit/creo2urdf/blob/master/README.md#installation-from-sources). Am I right?

-> Can you compile the plugin from sources?

I cannot follow [the instructions](https://github.com/icub-tech-iit/creo2urdf/blob/master/README.md#build-creo2urdf).  I don't get it. Are there any detailed instructions?

mfussi66, 2025-04-17:
> -> Did you make sure to install the c++ object toolkit ?
> 
> It was needed for the[ installation from source](https://github.com/icub-tech-iit/creo2urdf/blob/master/README.md#installation-from-sources). Am I right?

Actually I think it is needed even when you aim to use the precompiled .dll, see the note in the readme

> The provided dll has not been unlocked, therefore the Toolkit licence is required for using it . 

We can clarify that we are referring to the C++ object toolkit to avoid future doubts.

> By the way, I followed [this](https://learn.microsoft.com/en-us/vcpkg/get_started/get-started-vscode?pivots=shell-powershell) from VSCode to make sure I have it correctly.
> 
> -> Can you compile the plugin from sources?
> 
> I am not sure if I follow [the instructions](https://github.com/icub-tech-iit/creo2urdf/blob/master/README.md#build-creo2urdf) correctly. That's why I used the compiled version you provided. I don't get it. Are there any detailed instructions?

For compilation you need the Visual Studio Compiler, that you can install with the Visual Studio IDE, probably the community edition will work as well. 2019 and 2022 should work fine

CyrusTZ, 2025-04-17:
I already have Visual Studio Community 2022 installed. I checked if the C++ compiler, MinGW64, and VCPKG are installed correctly, which are existed.

Could you please expand the details on where to export and pass the requirements mentioned [here](https://github.com/icub-tech-iit/creo2urdf/blob/master/README.md#build-creo2urdf)?

Thanks