mesh-iit/creo2urdf #136: Add support for spherical joints

Nicogene, 2025-06-11:
Right now here is the list of type of joints supported:
- Revolute
- Fixed
- Prismatic

It would be nice to add the support for the spherical joints (cc @FabioBergonti @valegagge @maggia80)

We need to:
- On creo2urdf side check how they are defined at the `ElementTree` level
- Add the support of this type of joints in `idyntree` (cc @traversaro)

traversaro, 2025-06-26:
Hello @xela-95 @fabiobergonti @carlottasartore @giulioromualdi @Giulero, 

the URDF does not support ball/spherical joints out of the box, see https://chatgpt.com/share/685d1830-bb10-8006-a9f5-ff840a8cf3ba for a related ChatGPT convo. So the question is, how do you want to export this info? 

Possible options:

## iDynTree specific-extension

Export a joint named:

~~~
<joint name="my_joint_revolute" type="ball"> 
    <origin xyz="0 0 1" rpy="0 0 3.1416"/>
    <parent link="link1"/>
    <child link="link2"/>
</joint>
~~~

Pro:
* Simple

Cons:
* Ignored by any other non-iDynTree URDF parser

## Export as three consecutive revolute

Export:

~~~
<joint name="my_joint_revolute_x" type="revolute"> 
    <origin xyz=". . ." rpy=". . ."/>
    <parent link="link1"/>
    <child link="my_joint_revolute_dummy_body_1"/>
</joint>

<joint name="my_joint_revolute_y" type="revolute"> 
    <origin xyz=". . ." rpy=". . ."/>
    <parent link="my_joint_revolute_dummy_body_1"/>
    <child link="my_joint_revolute_dummy_body_2"/>
</joint>

<joint name="my_joint_revolute_y" type="revolute"> 
    <origin xyz=". . ." rpy=". . ."/>
    <parent link="my_joint_revolute_dummy_body_2"/>
    <child link="link2"/>
</joint>
~~~

Pro:
* Any URDF importer supports it

Con:
* The URDF importer needs to have some special logic to convert this sequence back to a spherical joint (o3 mentioned that the Pinocchio parser is able to automatically convert them back to spherical joints that is a nice feature, but we are not sure if that is an hallucination or not)

CarlottaSartore, 2025-06-26:
What we are currently doing for loading the model in Jaxsim  is loading a URDF with a fixed joint and then imposing a connect constraint in Jaxsim (thus closing the kinematic chain and implementing a passive spherical joint).
We have also started to develop some utilities that convert 3 consecutive revolute joints into a fixed one and vice versa (see https://github.com/ami-iit/element_differentiable_simulators_for_codesign/issues/12#issuecomment-2988398971),

 So I agree that  O2  seems the best option

FabioBergonti, 2025-06-26:
O2 looks good to me too. Quick question: how do we handle inertial parameters for the dummy links? Using very small values might cause instabilities in simulation, right?

CarlottaSartore, 2025-06-26:
It depends on how the simulation handles dummy link ( without inertial parameters ) it could be that they are able to identify them . For instance old gazebo is not able but jaxsim converts them to frames and I think also isaac sim have something to deal with massless link. 
Anyhow in the transformation to spherical joint ( as I think you will also do for Isaac sim) the dummy lins should be removed in favor of the spherical joint

traversaro, 2025-06-26:
Yes, I agree with @CarlottaSartore . I think we need some post-processing logic in the URDF --> USD (IsaacSim), URDF --> SDF (JaxSim/gz sim) or eventually URDF --> MJCF (MuJoCo, this is something that could be include in https://github.com/ami-iit/mujoco-urdf-loader) to post-process loaded models and convert the revolute joints back to sphericals, removign the dummy link at all. For IsaacSim it would be cool to have a `openusd-urdf-loader` library to contain this kind of functionality, but for now it can be something hardcoded in any project that loads the URDF.

closed 2025-10-24 by Nicogene

Nicogene, 2025-10-24:
Hi!

The support of the spherical joint has been released:
- https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.5.15

<blockquote>[image]<div>[image] GitHub</div><div><strong><a href="/icub-tech-iit/creo2urdf/releases/tag/v0.5.15">Release v0.5.15 · icub-tech-iit/creo2urdf</a></strong></div><div>What's Changed

Add Support for spherical joint by @Nicogene in #152

Full Changelog: v0.5.14...v0.5.15</div></blockquote>