mesh-iit/creo2urdf #140: 'Root' option does not work when exportedFrames are rigidly connected to it

FabioBergonti, 2025-06-19:
Hi!
I'm exporting an assembly with two `exportedFrames` (`l_ankle_spherical_parent_1` and `l_ankle_spherical_parent_2`) that are rigidly connected to the intended root link (`l_lower_leg`). However, when the URDF is generated, the first link is unexpectedly set to one of the exported frames (`l_ankle_spherical_parent_1`), instead of the one I selected in the yaml (`l_lower_leg`).

Here is an excerpt of the generated URDF:

```xml
<?xml version="1.0"?>
<robot name="test">
  <link name="l_ankle_spherical_parent_1"/>
  <joint name="l_ankle_spherical_parent_1_fixed_joint" type="fixed">
    <origin xyz="0 -0.043000000000000003 1.1389215232070394e-17" rpy="0 -0 0"/>
    <parent link="l_ankle_spherical_parent_1"/>
    <child link="l_lower_leg"/>
  </joint>
```

If I remove the `exportedFrames`, the selected root link (`l_lower_leg`) is correctly used as the root in the URDF, as expected.

I've attached [the files used for both tests](https://github.com/user-attachments/files/20825420/files.zip) for your reference.

Tested with:

* Creo 9.0.2.0
* [creo2urdf v0.5.11](https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.5.11)

Let me know if I can provide more details to help reproduce the issue.

Nicogene, 2025-06-20:
Ho @FabioBergonti! Can you post also the yaml ? I will try to reproduce it with the two bars assembly

traversaro, 2025-06-20:
This sounds related to the setting of the [`iDynTree::ModelExporterOptions::exportFirstBaseLinkAdditionalFrameAsFakeURDFBase`](https://robotology.github.io/idyntree/structiDynTree_1_1ModelExporterOptions.html#a776c8f98ae5c58e038e3a1962f9fd318) parameter, that was an old workaround for a still present issue in KDL/ROS (https://github.com/ros/kdl_parser/issues/27, https://github.com/ros/robot_model/issues/6), see the issue https://github.com/robotology/idyntree/issues/1201 . 

As this seems confusing in the `creo2urdf` world, probably we can just set it to `false` in `creo2urdf` ?

closed 2025-06-20 by Nicogene

FabioBergonti, 2025-06-20:
I tested the new release https://github.com/icub-tech-iit/creo2urdf/releases/tag/v0.5.12 and the problem disappeared! 
Thanks @Nicogene @traversaro!